renode/renode #86: HiDPI monitor support

DurandA, 2020-03-29:
The Monitor (Renode CLI) starts with a tiny font on my HiDPI setup. I am running Ubuntu 18.04.

Is there a way to adapt the font size? If not, adding proper HiDPI support would be great.

PiotrZierhoffer, 2020-03-29:
Thank you for the report.

The font size can be changed in two ways:

- by pressing Ctrl+Shit+Plus/Minus Sign
- by adjusting your Renode config - look for it in ~/.config/renode/config, option `font-size`

The second option is, of course, persistent and allows you to adjust the font face as well.

Please let me know if it solves your problem.

PiotrZierhoffer, 2020-03-29:
The reason is Windows :) It lacked proper vt100 terminal, so we made one. FYI, in the Renode config you can also change Termsharp to GnomeTerminal, if you prefer.