renode/renode #88: Acess simulator uart console

tamnguyenchi93, 2020-04-15:
I try to run renode without gui.
> renode --disable-xwt

In another console I use telnet to connect :
```
$ telnet localhost 1234
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Renode, version 1.8.2.22285 (dc338ac5-201911131222)

(monitor) 
```
Inside of monitor console:
```
include @scripts/single-node/sifive_fe310.resc
sysbus LoadELF @/workspace/zephyr/build/console/zephyr/zephyr.elf
start
```
Which elf file is build from zephyr for sifive board:
https://github.com/zephyrproject-rtos/zephyr/issues/23395

After start I see renode log:
```bash
02:28:52.0431 [INFO] Loaded monitor commands from: /opt/renode/./scripts/monitor.py
02:28:52.0669 [INFO] Monitor available in telnet mode on port 1234
02:32:46.7843 [INFO] Including script: /opt/renode/scripts/single-node/sifive_fe310.resc
02:32:46.8009 [INFO] System bus created.
02:32:47.4167 [DEBUG] Segment size automatically calculated to value 64KiB (3)
02:32:47.5460 [DEBUG] Segment size automatically calculated to value 16MiB
02:32:47.7632 [DEBUG] sysbus: Loading ELF /tmp/renode-9955/cc049b2c-5745-4ff9-bc99-fcfa18df32d3.tmp.
02:32:47.7704 [INFO] sysbus: Loading segment of 20520 bytes length at 0x20400000.
02:32:47.7874 [DEBUG] sysbus: Segment loaded.
02:32:47.7876 [INFO] sysbus: Loading segment of 324 bytes length at 0x20405028.
02:32:47.7877 [DEBUG] sysbus: Segment loaded.
02:32:47.7878 [INFO] sysbus: Loading segment of 7944 bytes length at 0x80000148.
02:32:47.7879 [DEBUG] sysbus: Segment loaded.
02:32:47.8208 [INFO] cpu: Setting PC value to 0x20401BC4.
02:32:56.6377 [DEBUG] sysbus: Loading ELF /workspace/zephyr/build/console/zephyr/zephyr.elf.
02:32:56.6426 [INFO] sysbus: Loading segment of 18748 bytes length at 0x20400000.
02:32:56.6427 [DEBUG] sysbus: Segment loaded.
02:32:56.6428 [INFO] sysbus: Loading segment of 348 bytes length at 0x2040493C.
02:32:56.6429 [DEBUG] sysbus: Segment loaded.
02:32:56.6430 [INFO] sysbus: Loading segment of 4576 bytes length at 0x80000160.
02:32:56.6431 [DEBUG] sysbus: Segment loaded.
02:32:56.6584 [INFO] cpu: Setting PC value to 0x20400000.
02:33:01.8059 [INFO] SiFive-FE310: Machine started.
02:33:01.9051 [WARNING] plic: Unhandled write to offset 0x200000, value 0x0.
02:33:01.9055 [WARNING] gpioInputs: Unhandled write to offset 0x38, value 0x0.
02:33:01.9090 [WARNING] sysbus: [cpu: 0x20400FAC] (tag: 'PRCI_PLLCFG') WriteDoubleWord to non existing peripheral at 0x10008008, value 0x60000.
02:33:01.9092 [WARNING] sysbus: [cpu: 0x20400FB4] WriteDoubleWord to non existing peripheral at 0x1000800C, value 0x100.
02:33:01.9099 [WARNING] sysbus: [cpu: 0x20400FB8] (tag: 'PRCI_PLLCFG') ReadDoubleWord from non existing peripheral at 0x10008008, returning 0xFFFFFFFF.
02:33:01.9100 [WARNING] sysbus: [cpu: 0x20400FC8] (tag: 'PRCI_PLLCFG') WriteDoubleWord to non existing peripheral at 0x10008008, value 0xFFFFFFFF.
02:33:01.9101 [WARNING] sysbus: [cpu: 0x20400FCC] (tag: 'PRCI_HFROSCCFG') ReadDoubleWord from non existing peripheral at 0x10008000, returning 0xFFFFFFFF.
02:33:01.9103 [WARNING] sysbus: [cpu: 0x20400FDC] (tag: 'PRCI_HFROSCCFG') WriteDoubleWord to non existing peripheral at 0x10008000, value 0xBFFFFFFF.
02:33:01.9107 [WARNING] gpioInputs: Unhandled read from offset 0x3C.
02:33:01.9108 [WARNING] gpioInputs: Unhandled write to offset 0x3C, value 0x0.
02:33:01.9109 [WARNING] gpioInputs: Unhandled read from offset 0x38.
02:33:01.9110 [WARNING] gpioInputs: Unhandled write to offset 0x38, value 0x10000.
02:33:01.9111 [WARNING] gpioInputs: Unhandled read from offset 0x3C.
02:33:01.9112 [WARNING] gpioInputs: Unhandled write to offset 0x3C, value 0x0.
02:33:01.9113 [WARNING] gpioInputs: Unhandled read from offset 0x38.
02:33:01.9114 [WARNING] gpioInputs: Unhandled write to offset 0x38, value 0x20000.
02:33:01.9115 [WARNING] gpioInputs: Unhandled read from offset 0x3C.
02:33:01.9116 [WARNING] gpioInputs: Unhandled write to offset 0x3C, value 0x0.
02:33:01.9117 [WARNING] gpioInputs: Unhandled read from offset 0x38.
02:33:01.9118 [WARNING] gpioInputs: Unhandled write to offset 0x38, value 0x4.
02:33:01.9119 [WARNING] gpioInputs: Unhandled read from offset 0x3C.
02:33:01.9120 [WARNING] gpioInputs: Unhandled write to offset 0x3C, value 0x0.
02:33:01.9121 [WARNING] gpioInputs: Unhandled read from offset 0x38.
02:33:01.9122 [WARNING] gpioInputs: Unhandled write to offset 0x38, value 0x8.
02:33:01.9123 [WARNING] gpioInputs: Unhandled read from offset 0x3C.
02:33:01.9124 [WARNING] gpioInputs: Unhandled write to offset 0x3C, value 0x0.
02:33:01.9125 [WARNING] gpioInputs: Unhandled read from offset 0x38.
02:33:01.9126 [WARNING] gpioInputs: Unhandled write to offset 0x38, value 0x10.
02:33:01.9127 [WARNING] gpioInputs: Unhandled read from offset 0x3C.
02:33:01.9128 [WARNING] gpioInputs: Unhandled write to offset 0x3C, value 0x0.
02:33:01.9129 [WARNING] gpioInputs: Unhandled read from offset 0x38.
02:33:01.9130 [WARNING] gpioInputs: Unhandled write to offset 0x38, value 0x20.
02:33:01.9131 [WARNING] gpioInputs: Unhandled read from offset 0x3C.
02:33:01.9132 [WARNING] gpioInputs: Unhandled write to offset 0x3C, value 0x0.
02:33:01.9133 [WARNING] gpioInputs: Unhandled read from offset 0x38.
02:33:01.9134 [WARNING] gpioInputs: Unhandled write to offset 0x38, value 0x200.
02:33:01.9135 [WARNING] gpioInputs: Unhandled read from offset 0x3C.
02:33:01.9136 [WARNING] gpioInputs: Unhandled write to offset 0x3C, value 0x0.
02:33:01.9137 [WARNING] gpioInputs: Unhandled read from offset 0x38.
02:33:01.9138 [WARNING] gpioInputs: Unhandled write to offset 0x38, value 0x400.
02:33:01.9207 [WARNING] gpioInputs: Unhandled write to offset 0x10, value 0x0.
02:33:01.9208 [WARNING] gpioInputs: Unhandled write to offset 0x18, value 0x0.
02:33:01.9209 [WARNING] gpioInputs: Unhandled write to offset 0x20, value 0x0.
02:33:01.9211 [WARNING] gpioInputs: Unhandled write to offset 0x28, value 0x0.
02:33:01.9212 [WARNING] gpioInputs: Unhandled write to offset 0x30, value 0x0.
02:33:01.9213 [WARNING] gpioInputs: Unhandled write to offset 0x40, value 0x0.
02:33:01.9222 [WARNING] sysbus: [cpu: 0x20402628] ReadDoubleWord from non existing peripheral at 0x10034060.
02:33:01.9223 [WARNING] sysbus: [cpu: 0x20402630] WriteDoubleWord to non existing peripheral at 0x10034060, value 0x0.
02:33:01.9230 [WARNING] sysbus: [cpu: 0x20402628] ReadDoubleWord from non existing peripheral at 0x10024060.
02:33:01.9231 [WARNING] sysbus: [cpu: 0x20402630] WriteDoubleWord to non existing peripheral at 0x10024060, value 0x0.
02:33:01.9269 [DEBUG] uart0: [+14.28s host +0s virt 0s virt from start] *** Booting Zephyr OS build zephyr-v2.2.0-1431-g8b6930ebef7d  ***
02:33:01.9297 [DEBUG] uart0: [+3.61ms host +0s virt 0s virt from start]   You should see another line with instructions below. If not,
02:33:01.9303 [DEBUG] uart0: [+0.62ms host +0s virt 0s virt from start]   the (interrupt-driven) console device doesn't work as expected:
02:33:01.9556 [DEBUG] uart0: [+25.27ms host +0.1ms virt 0.1ms virt from start]   Start typing characters to see them echoed back
```
But I dont know how to attach to simulator console. As document with GUI that new windows of simulator console will open.
So How can I access to simulator console without GUI?

PiotrZierhoffer, 2020-04-15:
The simplest option would be to access the UART via telnet as well. In monitor type this before running "start":

```
emulation CreateServerSocketTerminal 4567 "uart-con" # this creates a connection listening on port 4567
connector Connect uart0 uart-con # this connects the uart to the connection you just created
```

tamnguyenchi93, 2020-04-16:
@PiotrZierhoffer Thank you for your help. I did it.

closed 2020-04-16 by tamnguyenchi93

mithro, 2020-04-16:
@PiotrZierhoffer / @mgielda -- Maybe the renode docs could have a "using renode without a GUI" type section? I think that would be interesting to lots of people.