renode/renode #89: Creating binary dump of sysbus.uart1 to file

gustavosinbandera1, 2020-04-24:
Hi guys it's possible to dump the data that is written to the uart1 as a file? I've been
looking but i've only found how to dump logs, but I need the binary data to do some tests on it.

PiotrZierhoffer, 2020-04-27:
Hi

Right now you can approach it via a socket and your own scripting:

```
emulation CreateServerSocketTerminal 3333 "backend" 
connector Connect sysbus.uart1 backend
```

Please note that we are working on a file backend - it should be available in a week or two.

gustavosinbandera1, 2020-04-27:
Thanks for you reply, in this moment was possible to me get the goal, doing a new peripheral and put  code to open ports and write files, its no the best code but works, now I am going  to do the test with the suggested solution. https://gist.github.com/gustavosinbandera1/75b60fc25ff7bb513eb42148be566134

closed 2020-04-27 by gustavosinbandera1

mateusz-holenko, 2020-04-29:
Hi @gustavosinbandera1,

I'm writing to let you know that `UartFileBackend` has been just merged to Renode master: https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Extensions/Utilities/UartFileBackend.cs. 

You can use it to dump UART content to a file - just execute the following command in monitor:
```
(machine-0) uart CreateFileBackend @/tmp/uart.dump
```
(add `true` at the end if you want to disable buffering and flush each character right away - this might decrease performance though).

Hope it'll be useful to you!

gustavosinbandera1, 2020-04-29:
Thanks, I appreciate you take your time to help me