renode/renode #91: Help with crash debugging

wojteklewandowski-pl, 2020-04-30:
Hi, I am adding custom peripheral to the renode, but this peripheral requires Linux shared C/C++. I have created the C# wrapper for it using pinvoke calls and separately, without renode it works. 
When I have added it to the renode I have a strange crash. 
First of all, I am creating deb package and installing it on the OS. It may be a possible reason, because maybe when it is installed using root privileges it does not see my library? I have copied it to /opt/renode/bin and have it on LD_LIBRARY_PATH. So If there

About the crash. When I load a platform description with my device, and it calls the constructor form C/C++ library it shows:
`2020-04-30 08:25:26: Internal Error: Exception handling process abandoned.
This happens when a handler cannot be found for a thrown exception,
or for some other exceptional circumstance that makes impossible
to continue the handling process.
Please, contact Technical Support.
See full backtrace in /opt/renode/backtrace.log.
2020-04-30 08:25:28: Internal Error: Exception handling process abandoned.
This happens when a handler cannot be found for a thrown exception,
or for some other exceptional circumstance that makes it impossible
to continue the handling process.
Please, contact Technical Support.
See full backtrace in /opt/renode/backtrace.log.
`

But there is no backtrace in such a place, so I don't know what to check.
Maybe someone has an idea how to solve it. My guess is that it is a problem with renode installed as root. Maybe it sees libraries/paths/environment a bit different so one possible solution would be to build it not as a package, but it looks the only possible different way is to build the portable but there is liking problem when building portable

` /usr/lib/pkgconfig/../../lib/libmono-2.0.a(llvm-runtime.o): In function 
`mono_llvm_cpp_catch_exception':
(.text+0x43): undefined reference to `__cxa_begin_catch'
(.text+0x4f): undefined reference to `__cxa_end_catch'
/usr/lib/pkgconfig/../../lib/libmono-2.0.a(llvm-runtime.o):(.data.rel.local.DW.ref._ZTIPi[DW.ref._ZTIPi]+0x0): undefined reference to `typeinfo for int*'
/usr/lib/pkgconfig/../../lib/libmono-2.0.a(llvm-runtime.o):(.data.rel.local.DW.ref.__gxx_personality_v0[DW.ref.__gxx_personality_v0]+0x0): undefined reference to `__gxx_personality_v0'
collect2: error: ld returned 1 exit status`

wojteklewandowski-pl, 2020-04-30:
I have confirmed that it works as root when I add a path to my library to the LD_LIBRARY_PATH. I don't know why it does not work without root. I have the same LD_LIBRARY_PATH as when I test as root

PiotrZierhoffer, 2020-05-27:
Just to clarify, by "works as root" you mean when it's installed via deb, but run as a user?

This would probably be the reason why `/opt/renode/backtrace.log` does not get created - the user does not have the rights. Running Renode with `sudo` should allow the backtrace to be created, but it's obviously not "the way".

My understanding is that `LD_LIBRARY_PATH` should work fine if set correctly, regardless of the user running the software. 

Regarding portable builds - we abuse mono quite substantially here, using a very unstable API. That's why it needs a very specific mono version to compile properly. It works on `5.20.1.19`

wojteklewandowski-pl, 2020-05-27:
Hi Piotr, thank you for the answer. I have installed renode as root (via sudo). Then ran Renode as a user and it crashes when trying to pinvoke to my C/C++ Library (I have LD_LIBRARY_PATH set correctly). When I log as root, not as a user, and set the LD_LIBRARY_PATH to the same value I had as a user then everything works. Renode communicates with my library, so there is nor reason for backtrace to be created. It looks like calling pinvoke from renode to C/C++ library requires root privileges. The library works without root privileges when using it from C/C++

PiotrZierhoffer, 2020-05-27:
The point is we do tons of calls to C - cores are implemented in C, so there's a lot of back-and-forth. We don't use P/Invoke directly, but have our own mechanism for that (`NativeBinder.cs`).

Mono website https://www.mono-project.com/docs/advanced/pinvoke/#troubleshooting suggest calling the failing app with these env variables, maybe it will show something:
`MONO_LOG_LEVEL="debug" MONO_LOG_MASK="dll"`

PiotrZierhoffer, 2021-05-25:
I will close this one, as it got a bit stale. If the issue persists, feel free to reopen and please post the info I mentioned above

closed 2021-05-25 by PiotrZierhoffer