renode/renode #92: Zephyr stm32f4_disco Hello World samples does not seem to run correctly

vla3913, 2020-05-03:
Hello et thanks for this wonderful project.

I was just trying to run the Zephyr Hello World sample for several boards.
I get the expected result on HiFive1 and stm32f103 but not on stm32f4_discovery.

Steps to reproduce:

On Zephyr build environment:
```
 west build -p auto -b stm32f4_disco hello_world
```

In renode terminal:
```
include @scripts/single-node/stm32f4_discovery.resc
sysbus LoadElf @[my binary]
s
```

No "Hello world!" in the board terminal but i get the following errors from Renode:
```
23:15:28.3513 [WARNING] sysbus: [cpu: 0x8000B60] (tag: 'RCC_CFGR') ReadDoubleWord from non existing peripheral at 0x40023808, returning 0x19940A. (4166)
23:15:28.6266 [WARNING] sysbus: [cpu: 0x8000B5E] (tag: 'RCC_CFGR') ReadDoubleWord from non existing peripheral at 0x40023808, returning 0x19940A. (4167)
23:15:28.9008 [WARNING] sysbus: [cpu: 0x8000B64] (tag: 'RCC_CFGR') ReadDoubleWord from non existing peripheral at 0x40023808, returning 0x19940A. (3476)
```

Does it mean that `stm32f4_disco` label does not match the same board in Zephyr and Renode ?

georgedlu, 2020-05-04:
I have gotten Zephyr sample to work on stm32f4_disco by appending following rcc definition to platforms/boards/stm32f4_discovery.repl

rcc: Python.PythonPeripheral @ sysbus 0x40023800
    size: 0x400
    initable: true
    filename: "scripts/pydev/flipflop.py"

In stm32f4_discovery.resc you need to have:
showAnalyzer sysbus.uart2

Below is output on uart2 window when running zephyr built elf from samples/cpp_synchronization:

*** Booting Zephyr OS build zephyr-v2.2.0-1661-g2c9ae5f88143  ***
Create semaphore 0x20000090
Create semaphore 0x2000007c
main: Hello World!
coop_thread_entry: Hello World!
main: Hello World!
coop_thread_entry: Hello World!

vla3913, 2020-05-05:
Thanks @georgedlu, your solution is indeed working.
I understand that you declare a 0x400 memory space to emulate the registers the system tries to access to. Do you have a more detailed analysis of the issue ?
Can it be considered as a bug or is it a misuse on my part ?

georgedlu, 2020-05-05:
I was trying to get STM32F4 and STM32F7 models to work in Renode. I eventually noticed some subtle peripheral definitions errors by comparing with the reference manual from ST against the devicetree for the target boards in Zephyr. stm32f7_discovery-bb.repl has RCC declaration entry, though the size was only 0x4. I verified it on memory map in STM32F7 reference manual, Table 1 on page 70 shows RCC registers should be in 0x4002 3800 - 0x4002 3BFF. I fixed the size error by added two trailing zeroes.

For STM32F4, I didn't see RCC declaration at all. I checked the memory map in [STM32F410 reference manual](https://www.st.com/resource/en/reference_manual/dm00180366-stm32f410-advanced-armbased-32bit-mcus-stmicroelectronics.pdf). Table 1 on page 39 shows RCC at the same memory location, so I simply copied the fixed RCC entry from my stm32f7_discovery-bb.repl.

I guess these are bugs in the platform description files. I had fixed them recently, happy to share the resolution.

PiotrZierhoffer, 2020-05-27:
Hey @georgedlu and @vla3913 .

Indeed, although the repl files are supposed to be 1:1 with the platform itself, there might be bugs in them, or simplifications. Using Python peripherals for things like clock control, memory controllers etc is common in Renode, as they are usually not that interesting from the whole platform perspective. We tend to implement what we need, and flipflop.py is very often a solution for things we just want to have for the sake of initialization.

If you have any updates to share, please make a pull request, we'd gladly incorporate it.

vla3913, 2020-06-17:
@PiotrZierhoffer , I created a branch containing the fix.
It seems that I do not have the credentials to push the branch on renode/renode.git.
Do I have to fork the repository first to create the pull request ?

PiotrZierhoffer, 2020-06-17:
Yes. The procedure is as follows:
- read the Contributing Guidelines: https://github.com/renode/renode/blob/master/CONTRIBUTING.rst
- fork
- push a branch
- do a pull request
- accept the CLA
- then we review and either merge or ask for some improvements

vla3913, 2020-06-17:
> Yes. The procedure is as follows:
> 
>     * read the Contributing Guidelines: https://github.com/renode/renode/blob/master/CONTRIBUTING.rst
> 
>     * fork
> 
>     * push a branch
> 
>     * do a pull request
> 
>     * accept the CLA
> 
>     * then we review and either merge or ask for some improvements

Thanks. Pull request created : https://github.com/renode/renode/pull/101

PiotrZierhoffer, 2020-07-02:
And it is now merged to master, so I'm closing the issue

closed 2020-07-02 by PiotrZierhoffer