renode/renode #93: Any guide to add a off the shelf cortex-M MCU?

haragg2, 2020-05-11:
I am trying to add a Texas Instruments Tm4c1294ncpdt MCU. I have the svd file of the MCU which renode seems to use.

PiotrZierhoffer, 2020-05-27:
Renode uses SVD files for logging purposes only.

To add a new platform you need to:

1. prepare models of peripherals. In Renode we keep them here: https://dev.antmicro.com/git/repositories/renode-infrastructure/tree/master/src/Emulator/Peripherals/Peripherals
Remember to verify if we don't have the ones you need already.
We do not have developer documentation yet, but you can look at other similar models and adjust them to your needs.
2. prepare a platform description: https://github.com/renode/renode/tree/master/platforms/cpus. The documentation for REPL files is here: https://renode.readthedocs.io/en/latest/advanced/platform_description_format.html

haragg2, 2020-05-27:
Thanks @PiotrZierhoffer I will look into this.

closed 2020-05-27 by haragg2