renode/renode #94: Having issue with running FreeRTOS example sketch

Notorious811, 2020-05-16:
`#include <STM32FreeRTOS.h>

void Task1( void *pvParameters );

void setup() {
  Serial.begin(57600);
  while (!Serial) {}
  Serial.println("yo");

  xTaskCreate(
    Task1
    ,  (const portCHAR *)"print"   // A name just for humans
    ,  128  // This stack size can be checked & adjusted by reading the Stack Highwater
    ,  NULL
    ,  1  // Priority, with 3 (configMAX_PRIORITIES - 1) being the highest, and 0 being the lowest.
    ,  NULL );

  Serial.println("sup");
  vTaskStartScheduler();
  Serial.println("Insufficient RAM");
  while(1);
}
void loop()
{
  // Empty. Things are done in Tasks.
}

/*--------------------------------------------------*/
/*---------------------- Tasks ---------------------*/
/*--------------------------------------------------*/

void Task1(void *pvParameters)  // This is a task.
{
 (void) pvParameters;

  for (;;) // A Task shall never return or exit.
  {
    Serial.println("hello");  
    vTaskDelay( 1000 / portTICK_PERIOD_MS ); // wait for one second
  }
}`

I am trying to run this arduinosketch on renode emulator but i am not getting desired results. **FreeRTOS Task Scheduler** does not seems to be running properly or i am missing some configurations to run it properly. 

Above sketch have been compiled successfully using arduino ide with stm32duino stm32 boards support [https://github.com/stm32duino/Arduino_Core_STM32](url) ,i have taken care of building for the right board in arduino ide and made sure to start simulating on the same stm32F4-discovery board on renode emulator as well.

This is the .elf file of the above sketch:
[https://drive.google.com/open?id=1qmUdz6oMoZTkdCdV0Z5ZKmWzVdaYkj3Y](url)
[printsomething.ino.zip](https://github.com/renode/renode/files/4639215/printsomething.ino.zip)

I have used these sets of commands on renode emulator:
-mach create
-machine LoadPlatformDescription @platforms/boards/stm32f4_discovery-kit.repl
-sysbus LoadELF @Blink_AnalogRead.ino.elf 
-machine LoadPlatformDescription @add-ccm.repl
-showAnalyzer sysbus.uart2
-start

that's the output on sysbus.uart2 console:
[image]

I wanted to know why **FreeRTOS Task Scheduler** not running in renode? i should have getting stream of hello output in the uart2 console, please help

Notorious811, 2020-05-24:
This sketch is working on stm32 board, just not working on renode, guess renode cannot emulate RTOS

PiotrZierhoffer, 2020-05-27:
Can you post your repl file (add-ccm.repl) and the log output of Renode? I believe it might be related to #85 

Renode is in general software-agnostic, so it does not rely on any specific RTOS details

haragg2, 2020-05-27:
I was also facing the similar issue. It has to do with the priority bits implemented in any cortex-M based micro-controller and the way FreeRTOS figures out this value. It first stores the value 0xFF in one of the NVIC priority register and then read back the value. Ideally in STM32 case where 4 bits are implemented it should read back the value 0xF0 but on the renode it is reading back 0xFF (which means all 8 bits are implemented). This conflict with this macro configPRIO_BITS and config assert fails. I have hard coded this value to 0xF0 in the port.c file of FreeRTOS and everything starts working.

PiotrZierhoffer, 2020-05-27:
@haragg2 This is a very good hint, thank you. We'll look into this.

PiotrZierhoffer, 2020-05-28:
Until we get a proper fix for that in our platform description, you can create a repl that patches this value:

```
using "platforms/boards/stm32f4_discovery-kit.repl"

nvic:
    priorityMask: 0xF
```
and load this instead of the original repl.

haragg2, 2020-05-28:
@PiotrZierhoffer Is this `priorityMask` an internal renode variable?

PiotrZierhoffer, 2020-05-28:
@haragg2 it is a parameter for the NVIC constructor, by default equal to 0xFF: https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Cores/Arm-M/NVIC.cs#L27

Notorious811, 2020-05-29:
@PiotrZierhoffer @haragg2

Notorious811, 2020-05-29:
@haragg2 @PiotrZierhoffer thanks guys, it worked

@haragg2 
[image]

this is default configuration of NVIC priority bits for stm32F4discovery in arduino STM32 boards library.

[image]

this is default configuration of NVIC priority bits for FreeRTOS library in arduino FreeRTOS library for STM32.

@PiotrZierhoffer 

> Until we get a proper fix for that in our platform description, you can create a repl that patches this value:
> 
> ```
> using "platforms/boards/stm32f4_discovery-kit.repl"
> 
> nvic:
>     priorityMask: 0xF
> ```
> 
> and load this instead of the original repl.
> 
> I am not sure if it addresses @Notorious811 problem, but the one described by @haragg2 should be resolved this way.

it solved the issue, just need to change priorityMask: 0xF with priorityMask: 0xF0 to make it work, had to add 0 explicitly after 0xF.

closed 2020-06-08 by Notorious811

PiotrZierhoffer, 2020-06-19:
@Notorious811 we released the updated platforms just now.