renode/renode #95: How to send data to the UART peripheral of STM32F4 discovery kit board from outside the emulated environment?

haragg2, 2020-05-27:
I have been able to view the data on the console using showAnalyzer sysbus.uart2 but I want to send some data back on the UART that can be read by the board. Is there any capability to use a python program from which I can read and then write back on the UART? Basically I want to simulate a UART peripheral using this python program which talks to the board running on renode.

PiotrZierhoffer, 2020-05-28:
On Linux you can do the following:
```
emulation CreateServerSocketTerminal 1234 "uart-con"
connector Connect sysbus.uart0 uart-con
```
Then, you should be able to `telnet localhost 1234`.

If you need your terminal to be raw (probably more suitable for scripting), change the first command to
```
emulation CreateServerSocketTerminal 1234 "uart-con" false
```
This `false` will prevent Renode from sending terminal config (that enables, for example, automatic echo).

PiotrZierhoffer, 2020-05-28:
You can also create a `/dev/pts`-style file:
```
emulation CreateUartPtyTerminal "uart-con" @path/to/file
connector Connect sysbus.uart0 uart-con
```
This will create a file that will be a link to /dev/pts/X, which you can connect to. Please bear in mind that this is not a fully-fledged pts, so some ioctl calls run against it might fail.

haragg2, 2020-05-28:
Thanks @PiotrZierhoffer I will try this method.

closed 2020-06-07 by pgielda