renode/renode #96: Sensor ADXL345 samples data

tamnguyenchi93, 2020-05-29:
I try to run example that connect hifive_revb with ADXL345 sensor via I2c bus.

Here is my renode platform description:

```
maskRom: Memory.MappedMemory @sysbus 0x1000
    size: 0x1000

otp: Memory.MappedMemory @sysbus 0x20000
    size: 0x2000

dtim: Memory.MappedMemory @ sysbus 0x80000000
    size: 0x4000

uart0: UART.SiFive_UART @ sysbus 0x10013000
    IRQ -> plic@3

uart1: UART.SiFive_UART @ sysbus 0x10023000
    IRQ -> plic@4

cpu: CPU.RiscV32 @ sysbus
    cpuType: "rv32imac"
    privilegeArchitecture: PrivilegeArchitecture.Priv1_10
    timeProvider: clint

plic: IRQControllers.PlatformLevelInterruptController @ sysbus 0x0C000000
    3 -> cpu@11
    numberOfSources: 52

clint: IRQControllers.CoreLevelInterruptor  @ sysbus 0x02000000
    [0,1] -> cpu@[3,7]
    frequency: 62000000

qspi0Xip: Memory.MappedMemory @ sysbus 0x20000000
    size: 0x20000000

gpioInputs: GPIOPort.SiFive_GPIO @  sysbus 0x10012000
    [0-31] -> plic@[8-39]

i2c0: I2C.OpenCoresI2C @ sysbus 0x10016000
    // our model does not support interrupts yet, but if it did:
    // IRQ -> plic@52

sensor: Sensors.ADXL345 @ i2c0 0x53

sysbus:
    init:
        ApplySVD @https://dl.antmicro.com/projects/renode/svd/FE310.svd.gz
```
On firmware side I use zephyr OS.
When the firmware start. It try to get current sample count store at registor:
FifoStatus=0x39

But the value return is 0.
Show how can I initiate these samples.

I try to add 
> FeedSample(0xff, 0xee, 0xdd, 1);

in line function Reset at: 

https://github.com/renode/renode-infrastructure/blob/ca50c1f28b32c26d339757f8932e78b5f024205e/src/Emulator/Peripherals/Peripherals/Sensors/ADXL345.cs#L46

But it does not work. So I try to add
```csharp
            lock(samplesFifo)
            {
                FeedSample(0xff, 0xee, 0xdd, 1);
            }
```
in line https://github.com/renode/renode-infrastructure/blob/ca50c1f28b32c26d339757f8932e78b5f024205e/src/Emulator/Peripherals/Peripherals/Sensors/ADXL345.cs#L32
And I work. I wonder is that the right way to initiate sensor sample data?
And I see that there are function to feed data from file. How can I use it. Can I load with my own sample data in platform description or it has to happen before build step.

PiotrZierhoffer, 2020-06-08:
Hi

In general, you don't have to modify the code to add samples. The Renode CLI allows you to call these functions in runtime.

We recently published a demo of Zephyr + TFLite + LiteX/VexRiscv setup where we use this sensor for gesture detection: https://github.com/antmicro/litex-vexriscv-tensorflow-lite-demo

You can find the usage example here:
https://github.com/antmicro/litex-vexriscv-tensorflow-lite-demo/blob/master/renode/litex-vexriscv-tflite.resc

Sample files with data are in the same directory.

You'll be calling:
```
i2c0.sensor FeedSample 0xff 0xee 0xdd
```
(as the last `1` is a default value)

tamnguyenchi93, 2020-06-11:
@PiotrZierhoffer This is very nice.
Can we feed samples when the simulation runs? Or It have be done before it starts?

PiotrZierhoffer, 2020-06-11:
You can do it either before starting or while the simulation runs, it's not a problem

tamnguyenchi93, 2020-06-16:
Thanks you a lot. It is a nice effort.

closed 2020-06-16 by tamnguyenchi93