renode/renode #97: Timer is not working on STM32F4 discovery board

haragg2, 2020-05-29:
I have added the timer in the stm32f4 repl file using below commands
```timer2: Timers.STM32_Timer @ sysbus <0x40000000, +0x100>
    -> nvic@28```

But the timer is not working and the console prints a WARNING message `attempt to read Word from peripheral that doesn't support Word interface renode`

![Screenshot (34)](https://user-images.githubusercontent.com/10476471/83236786-b5d2dd00-a1b1-11ea-9530-a437a5c02bc6.png)

pgielda, 2020-06-07:
In general it means that the firmware you are running is trying to read a word (16-bit) value from a peripheral that is implemented as one that accepts only dword (32-bit) accesses.

What software are you using? Does it work on hardware?

PiotrZierhoffer, 2020-06-08:
If you are confident that this peripheral should allow word access, you can add `[AllowedTranslations(AllowedTranslation.WordToDoubleWord)]` in its model:

```
diff --git a/src/Emulator/Peripherals/Peripherals/Timers/STM32_Timer.cs b/src/Emulator/Peripherals/Peripherals/Timers/STM32_Timer.cs
index aa90094d8..f28aaaa9e 100644
--- a/src/Emulator/Peripherals/Peripherals/Timers/STM32_Timer.cs
+++ b/src/Emulator/Peripherals/Peripherals/Timers/STM32_Timer.cs
@@ -15,6 +15,7 @@
 
 namespace Antmicro.Renode.Peripherals.Timers
 {
+    [AllowedTranslations(AllowedTranslation.WordToDoubleWord)]
     public class STM32_Timer : LimitTimer, IDoubleWordPeripheral
     {
         public STM32_Timer(Machine machine) : base(machine.ClockSource, 10000000, direction: Direction.Ascending, limit: 0x100000, enabled: false) {
```

Please keep in mind that this particular model is a stub, and while it should generate interrupts as configured, it's not very robust.

haragg2, 2020-06-16:
@PiotrZierhoffer I have tried your suggestion of adding `[AllowedTranslations(AllowedTranslation.WordToDoubleWord)]` and build it. But still, it doesn't work.
[image]

Below are the logs when I added
`timer2: Timers.STM32_Timer @ sysbus <0x40000000, +0x100>`
`-> nvic@28`
[image]

Below are the logs when I removed the timer2 module from .repl file
[image]

What I noticed is that with no `timer2` module in the `.repl` file the timer module is recognized by the emulator. But in both cases, the timer is not working.
This is the simple timer setup.
[image]

marcelobarrosalmeida, 2020-07-14:
Hello

I am facing the same problem. 
I added timers 6 and 7 to stm32f4.repl, like below:

```
timer6: Timers.STM32_Timer @ sysbus <0x40001000, +0x400>
    -> nvic@54

timer7: Timers.STM32_Timer @ sysbus <0x40001400, +0x400>
    -> nvic@55
```
My code was generated by cubeIDE, using a sysclock of 42MHz. For timer 7 I used a preescaler of 41999 and a ARR of 999 for a 1s base time. I added interrupt handlers generation and priority as 1.

However, after calling `HAL_TIM_Base_Start_IT(&htim7);` I do not have any  timer interrupt and TIM7->CNT is not increasing. Generated files with modifications are below (I did not attached the entire HAL). Some fragments are below:

```
void HAL_TIM_PeriodElapsedCallback(TIM_HandleTypeDef *htim)
{
	if(htim->Instance == TIM7)
	{
		HAL_GPIO_TogglePin(LD10_GPIO_Port, LD10_Pin);
		printf("Led toggle\r\n");
	}
}

int main(void)
{
  /* Configure the system clock */
  SystemClock_Config();

  /* Initialize all configured peripherals */
  MX_GPIO_Init();
  MX_TIM7_Init();
  MX_USART1_UART_Init();
  MX_USART2_UART_Init();

  /* USER CODE BEGIN 2 */
  printf("Initialized\r\n");
  HAL_TIM_Base_Start_IT(&htim7);
  volatile uint32_t cnt = 0;
  /* USER CODE END 2 */

  /* Infinite loop */
  /* USER CODE BEGIN WHILE */
  while (1)
  {
	  cnt = __HAL_TIM_GET_COUNTER(&htim7);
	  printf("CNT = %lu\r\n",cnt);
	  HAL_Delay(1000);
    /* USER CODE END WHILE */

    /* USER CODE BEGIN 3 */
  }
  /* USER CODE END 3 */
}

```

I am new to renode and I can dig further with proper instructions. 

Thanks a lot.

[Core.zip](https://github.com/renode/renode/files/4920533/Core.zip)

mszprejda, 2020-10-22:
Hello,

@haragg2 and @marcelobarrosalmeida, thanks for reporting.

You can find an updated version of the STM32_Timer and related platform description files on master. Please take a look if it solves your problems.

grifcj, 2021-01-26:
Using a simple while loop, I was able to see timer interrupt for latest release (1.11.0.24832). I haven't quite got the expected interrupt frequency to line up though, i.e see the timer interrupt at 1 Hz as expected once 1 second of virtual time. I still need to check that hardware settings are correct for this, but I'm not sure if this is possible? The peripheral model would need to know clock frequency and increment count appropriately for every received quant?

[image]