renode/renode #98: AVR support

Turakar, 2020-05-30:
Hello,

first of all: Renode looks pretty cool. Sounds like a priori performance evaluation (i.e. How many MIPS do I need for this algorithm?) can be easily done with this tool. However, this is unrelated to this issue, just wanted to say thank you :smiley: 

Now to the real question: As far as I can tell from browsing the repositories, there is no AVR support present in Renode. I would therefore like to discuss the enhancement of Renode with AVR support. Questions I have include:

- Can it be done?
- Maybe it is even on a roadmap?
- What are the reasons not to do this?
- Is it a lot of / some / little work to implement this?
- What would be the steps for this? Guess this is something like implementing the architecture in `renode-infrastructure`, then creating some example boards with REPL configs.

The initial motivation comes from a project for a onboard-computer for a small satellite. Due to radiation, the robust AVRs are preferred for simple subtasks which do not need the computing power of a ARM architecture. This includes power management, some sensor drivers and others. These AVRs communicate with the central CPU, which could be a Cortex-M for example. To emulate this properly, it would be beneficial if all components could be emulated inside Renode. This would reduce the number of hardware iterations.

We are currently investigating ATmega8 for our project.

Looking forward to your thoughts on this topic!

pgielda, 2020-06-07:
Its true, we do not have AVR architecture. It obviously can be done, but in general AR is not really that popular anymore being slowly replaced by 32-bit architectures like ARM or RISC-V.
Having said I do not see any real obstacles in doing it, but its not in the roadmap for the reasons above.

Turakar, 2020-06-07:
Thanks for the clarification and I understand your reasoning - too bad the space industry is always some years behind due to the reliability and radiation constraints :|

closed 2020-06-07 by pgielda

septatrix, 2022-09-27:
I would not say that AVR is being replaced by 32-bit architectures. Other use cases emerge where ARM or RISC-V are preferred though the use cases for AVR still remain. As with programming languages it's always about the right tool for the job and sometimes 8-bit is more than sufficient and saving a few cents in millions of devices makes a large difference. The 8-bit and 16-bit markets are actually growing faster than 32-bit (and the majority of those are AVR). So AVR is still really popular and strong. And those use cases would still benefit from a tool like renode.

So I would still very much welcome support for this :)