renode/renode #99: JIT on Windows doesn't work well

AntonKrug, 2020-06-08:
I'm using Windows 10 64-bit  with .NET 4.8.03752 and current release of Renode.

When I use **block** body syntax of properties the model will compile JIT on windows just fine, here is simplified example of the model:

```
using Antmicro.Renode.Peripherals.Bus;

namespace Antmicro.Renode.Peripherals.Miscellaneous
{
    public class Simple : IDoubleWordPeripheral, IKnownSize
    {
        public long Size
        {
            get
            {
                return 10;
            }
        }

        public uint ReadDoubleWord(long offset)
        {
            return 0;
        }

        public void Reset()
        {
        }

        public void WriteDoubleWord(long offset, uint value)
        {
        }
    }
}
```

However when I write the same model with **expression** body syntax on the properties:

```
using Antmicro.Renode.Peripherals.Bus;

namespace Antmicro.Renode.Peripherals.Miscellaneous
{
    public class Simple : IDoubleWordPeripheral, IKnownSize
    {
        public long Size => 10;

        public uint ReadDoubleWord(long offset) => 0;

        public void Reset()
        {
        }

        public void WriteDoubleWord(long offset, uint value)
        {
        }
    }
}
```

Then the JIT compilation will fail:
```
Errors during compilation or loading:
There were compilation errors:
....\Simple.cs(8,26) : error CS1002: ; expected
....\Simple.cs(10,49) : error CS1002: ; expected
There was an error executing command 'machine LoadPlatformDescriptionFromString "simple: Miscellaneous.Simple @ sysbus 0x10100000"'
Error E04: Could not resolve type: 'Miscellaneous.Simple'.
At 1:9:
simple: Miscellaneous.Simple @ sysbus 0x10100000
        ^^^^^^^^^^^^^^^^^^^^^
```

Is there recommended and proven .NET version which should work with JIT on Windows all the time?

PiotrZierhoffer, 2020-06-08:
Hi Anton

This is a known limitation - the compiler API we use relies on older compilation mechanisms and does not allow for newer C# features, like expression-bodied members or inner methods.

It works the same on all OSes, so I allow myself to update the topic a bit.

We do want to update to newer API, especially to allow for compilation in the portable package, but we do not have any specifics at the moment. While the API changes are not very problematic here, the new mechanism will require significant changes to the build process, so it's not on the top of our list at the moment.

PiotrZierhoffer, 2020-06-10:
Actually, I will somewhat revert the title change - Mono is less strict about language version and compiles whatever it gets.

mateusz-holenko, 2021-05-05:
The C# compiler is now part of Renode - in the past we relied on the one installed in the system. Thanks to this we are now able to have the same experience on all platforms and in all packages (including the Linux portable one). As a bonus we support newer C# syntax as well, so with Renode 1.12 you should be able to dynamically include the problematic class.

closed 2021-05-05 by mateusz-holenko