renode/renode #102: Fails to build on Fedora 32

btashton, 2020-06-19:
I have tried to build the latest release on Fedora 32, but the build is failing here at what seems to be the local function definition.  It has been 10 years since I last wrote C# so it is very possible I am doing something wrong.

```
/home/bashton/nuttx/renode/src/Infrastructure/src/Emulator/Main/Core/Structure/MACAddress.cs(67,13): error CS1547: Keyword `void' cannot be used in this context
/home/bashton/nuttx/renode/src/Infrastructure/src/Emulator/Main/Core/Structure/MACAddress.cs(67,26): error CS1525: Unexpected symbol `(', expecting `,', `;', or `='
/home/bashton/nuttx/renode/src/Infrastructure/src/Emulator/Main/Core/Structure/MACAddress.cs(93,15): error CS1525: Unexpected symbol `MACAddress', expecting `class', `delegate', `enum', `interface', `partial', `ref', or `struct'
/home/bashton/nuttx/renode/src/Infrastructure/src/Emulator/Main/Core/Structure/MACAddress.cs(95,12): error CS1525: Unexpected symbol `var', expecting `class', `delegate', `enum', `interface', `partial', `ref', or `struct'
/home/bashton/nuttx/renode/src/Infrastructure/src/Emulator/Main/Core/Structure/MACAddress.cs(100,16): error CS1525: Unexpected symbol `if', expecting `class', `delegate', `enum', `interface', `partial', `ref', or `struct'
/home/bashton/nuttx/renode/src/Infrastructure/src/Emulator/Main/Core/Structure/MACAddress.cs(100,33): error CS1525: Unexpected symbol `(', expecting `class', `delegate', `enum', `interface', `partial', `ref', or `struct'
/home/bashton/nuttx/renode/src/Infrastructure/src/Emulator/Main/Core/Structure/MACAddress.cs(102,20): error CS1525: Unexpected symbol `if', expecting `class', `delegate', `enum', `interface', `partial', `ref', or `struct'
/home/bashton/nuttx/renode/src/Infrastructure/src/Emulator/Main/Core/Structure/MACAddress.cs(104,24): error CS1525: Unexpected symbol `throw', expecting `class', `delegate', `enum', `interface', `partial', `ref', or `struct'
/home/bashton/nuttx/renode/src/Infrastructure/src/Emulator/Main/Core/Structure/MACAddress.cs(104,33): error CS1514: Unexpected symbol `new', expecting `.' or `{'
/home/bashton/nuttx/renode/src/Infrastructure/src/Emulator/Main/Core/Structure/MACAddress.cs(104,31): error CS1530: Keyword `new' is not allowed on namespace elements
/home/bashton/nuttx/renode/src/Infrastructure/src/Emulator/Main/Core/Structure/MACAddress.cs(104,34): error CS1525: Unexpected symbol `OverflowException', expecting `class', `delegate', `enum', `interface', `partial', `ref', or `struct'
			Task "Csc" execution -- FAILED
			Done building target "CoreCompile" in project "/home/bashton/nuttx/renode/src/Infrastructure/src/Emulator/Main/Emulator.csproj".-- FAILED
		Done building project "/home/bashton/nuttx/renode/src/Infrastructure/src/Emulator/Main/Emulator.csproj".-- FAILED
	Task "MSBuild" execution -- FAILED
	Done building target "Build" in project "/home/bashton/nuttx/renode/Renode.sln".-- FAILED
Done building project "/home/bashton/nuttx/renode/Renode.sln".-- FAILED

Build FAILED.
Errors:
```

The version of Mono that I have:
```
❯ mono --version
Mono JIT compiler version 6.6.0.161 (tarball Wed Feb  5 07:34:41 UTC 2020)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
	TLS:           __thread
	SIGSEGV:       altstack
	Notifications: epoll
	Architecture:  amd64
	Disabled:      none
	Misc:          softdebug 
	Interpreter:   yes
	LLVM:          supported, not enabled.
	Suspend:       hybrid
	GC:            sgen (concurrent by default)
```

If there is anything more I can supply I am happy to.

PiotrZierhoffer, 2020-06-19:
Hi @btashton . I assume you are using the version of Mono bundled with Fedora. Unfortunately, it does not provide all the features we need. I would encourage you to follow these instructions: https://github.com/renode/renode#mononet

We'll update the docs to indicate that current Debian/Fedora bundled Mono releases are not good enough.

PiotrZierhoffer, 2020-07-02:
@btashton can you please retry? We are in the process of re-enabling standard mono bundled with different distros

mateusz-holenko, 2020-07-22:
Hi @btashton - just wanted to let you know that you should be able now to build the current Renode master branch using the default Mono package provided by Fedora.

btashton, 2020-07-23:
Hey sorry I did not get a chance to test this sooner.  Looks like this build gets a lot further along now, but I now hit this for each of the arch `obj-arm-32_le` `obj-ppc-32_le` `obj-i386-32_le` etc...
```
/home/bashton/renode/Renode.sln (default targets) ->
(Build target) ->
/home/bashton/renode/src/Infrastructure/src/Emulator/Main/Emulator.csproj (default targets) ->
/usr/lib/mono/xbuild/14.0/bin/Microsoft.CSharp.targets (CoreCompile target) ->

	/home/bashton/renode/src/Infrastructure/src/Emulator/Main/Peripherals/Bus/SVDParser.cs(541,29): warning CS0219: The variable `derivedRegister' is assigned but its value is never used

/home/bashton/renode/Renode.sln (default targets) ->
(Build target) ->
/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/cores-arm.csproj (default targets) ->
/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/cores.template.csproj (CompileTranslationLib target) ->
/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj (VerifyProperties;GenerateFlags;Compile;Link;Build) ->
(Link target) ->

	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/exec.o:(.bss+0x42460): multiple definition of `maximum_block_size'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x0): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/exec.o:(.bss+0x42468): multiple definition of `env'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x8): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/debug.o:(.bss+0x13880): multiple definition of `maximum_block_size'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x0): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/debug.o:(.bss+0x13888): multiple definition of `env'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x8): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/atomic.o:(.bss+0x0): multiple definition of `maximum_block_size'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x0): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/atomic.o:(.bss+0x8): multiple definition of `env'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x8): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/cpu-exec.o:(.bss+0x8): multiple definition of `env'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x8): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/cpu-exec.o:(.bss+0x4): multiple definition of `maximum_block_size'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x0): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/exports.o:(.bss+0xc): multiple definition of `maximum_block_size'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x0): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/exports.o:(.bss+0x10): multiple definition of `env'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x8): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/arch/translate-all.o:(.bss+0x0): multiple definition of `cpu_env'; obj/Release/obj-arm-32_le/tlib/exec.o:(.bss+0x0): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/arch/translate-all.o:(.bss+0x4): multiple definition of `maximum_block_size'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x0): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/arch/translate-all.o:(.bss+0x8): multiple definition of `env'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x8): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/arch/arm/translate.o:(.bss+0x0): multiple definition of `maximum_block_size'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x0): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/arch/arm/translate.o:(.bss+0x8): multiple definition of `env'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x8): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/tlib/arch/arm/helper.o:(.bss+0x0): multiple definition of `maximum_block_size'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x0): first defined here
[... cut ...]
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/renode/renode_callbacks.o:(.bss+0x98): multiple definition of `maximum_block_size'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x0): first defined here
	/home/bashton/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj:  warning : /usr/bin/ld: obj/Release/obj-arm-32_le/renode/renode_callbacks.o:(.bss+0xa0): multiple definition of `env'; obj/Release/obj-arm-32_le/tlib/helper.o:(.bss+0x8): first defined here
```

mateusz-holenko, 2020-07-24:
I was able to reproduce the problem and believe it's related to the version of GCC is used in Fedora. 

https://github.com/renode/renode/commit/63eee7c15e282184e62e1fbc7cf15f4c52b4f27f (already merged to master) fixes the problem for me - could you check if it works for you too, @btashton?

btashton, 2020-07-24:
I have not done much testing, but the build on master did pass.  Feel free to close this ticket unless there is more testing that you would like me to do.  Thanks for making the changes to get this to work, I normally would like to contribute more in the process, but mono is not something I really know much about.

mateusz-holenko, 2020-07-24:
I guess we can close this one - if there are any more problems with Renode on Fedora we should have a separate task for them.

Thanks for reporting the issue and testing the fix!

closed 2020-07-24 by mateusz-holenko