renode/renode #103: AIRCR SYSRESETREQ not implemented

franc0is, 2020-06-20:
A write to that bit should reset the system, provided VECTKEY is correct.

Currently it looks like only the PRIGROUP field in implemented: https://github.com/renode/renode-infrastructure/blob/b0b01348a3f17063a5e630de0e1a4c2279e98204/src/Emulator/Cores/Arm-M/NVIC.cs#L182

For my use case I would need registers to reset to their default value, and the core to start executing from the reset vector _without_ while RAM & ROM state is preserved.

franc0is, 2020-06-20:
Register definition: https://developer.arm.com/docs/100235/0004/the-cortexm33-peripherals/system-control-block/application-interrupt-and-reset-control-register

PiotrZierhoffer, 2020-07-02:
This is merged

closed 2020-07-02 by PiotrZierhoffer