renode/renode #104: STM32F4: Flash operations silently fail

franc0is, 2020-06-21:
Since the flash on STM32F4 is currently implemented as generic `MappedMemory`, all standard operations through the Flash CR silently fail. Attempting to write the Flash CR should trigger a more explicit warning log rather than the current debug log:
```
00:47:26.8846 [DEBUG] sysbus: Write value 0x80000000 to FLASH:CR (0x40023C10)
```
I suggest:
```
00:47:26.8846 [WARNING] sysbus: Write value 0x80000000 to FLASH:CR (0x40023C10) has no effect (Not Implemented)
```

PiotrZierhoffer, 2020-07-03:
We increased the verbosity of SVD-generated logs, and made them more explicit. STM32 flash is not currently on our roadmap, but I believe current logs will be more precise

franc0is, 2020-07-09:
Awesome, thanks!

Fixed in https://github.com/renode/renode-infrastructure/commit/fe0dcb83c34ed36c1552181c9db957b23934a4d9

closed 2020-07-09 by franc0is