renode/renode #105: Failed to execute MI command

Vishvas286, 2020-07-01:
I've written a simple C code and I was following this video: https://www.youtube.com/watch?v=SnbfRAZRTAc

When I hit Debug I get the following error:

```
Error in final launch sequence
Failed to execute MI command:
-target-select remote localhost:3333
Error message from debugger back end:
localhost:3333: No connection could be made because the target machine actively refused it.
Failed to execute MI command:
-target-select remote localhost:3333
Error message from debugger back end:
localhost:3333: No connection could be made because the target machine actively refused it.
localhost:3333: No connection could be made because the target machine actively refused it.
```

I'm new to SoftColsole, so could anyone help me out with this
Thank you

TommyMurphyTM1234, 2020-07-01:
SoftConsole v5.3 is very old.
You should use SoftConsole v6.2.

https://www.microsemi.com/product-directory/design-tools/4879-softconsole#downloads

Vishvas286, 2020-07-01:
@TommyMurphyTM1234 
I have installed SoftConsole 6.2, but I couldn't find a detailed enough guide to using Renode with it, so I went with that video
If you know about any tutorial or a guide for new comers, please let me know
Thanks

TommyMurphyTM1234, 2020-07-01:
Did you try the example debug/groiup launchers and the release notes?

Vishvas286, 2020-07-01:
@TommyMurphyTM1234 
Yes, I followed that first, got the same error and then searched for a YouTube video

AntonKrug, 2020-07-01:
@Vishvas286 I concur to use recent versions of the tools. The bundled release notes document has a lot of steps and common 'gotchas' described. Just from the glance it could be firewall or running 2 instances, or running wrong launcher, or something minor. Would you be able to record a video to show us your launchers settings, what you are launching prior the error?

ARAVINDAARONE, 2025-03-22:
Failed to execute MI command: target remote localhost:61239 Error message from debugger back end: localhost:61239: Connection timed out. localhost:61239: Connection timed out.

How to solve this issue?

TommyMurphyTM1234, 2025-03-22:
> Failed to execute MI command: target remote localhost:61239 Error message from debugger back end: localhost:61239: Connection timed out. localhost:61239: Connection timed out.
> 
> How to solve this issue?

Those Eclipse error messages don't really shed much light on the problem. Check the relevant terminal view for further messages and/or try to run Renode manually and standalone outside of the Eclipse IDE using the same command and options as the Eclipse launch configuration to see what the issue is.

ARAVINDAARONE, 2025-03-22:
i am new to these STM-32
can you brief it please?

TommyMurphyTM1234, 2025-03-22:
> i am new to these STM-32 can you brief it please?

I don't know what this means.
But this is not the appropriate forum to get a tutorial on how to use STM32 if that's what you mean.