renode/renode #106: Build fails silently due to user env var being set

tcal-x, 2020-07-02:
If the user has PLATFORM set to a value in their environment, then `./build.sh` will fail with no warning or error.

If the redirect to /dev/null is removed, the actual error can be seen:
```
$ export PLATFORM=FOOBAR
$ ./build.sh
Required Renode libraries already downloaded. To repeat the process remove .renode_libs_fetched file.

>>>> xbuild tool is deprecated and will be removed in future updates, use msbuild instead <<<<

XBuild Engine Version 14.0
Mono, Version 6.8.0.105
Copyright (C) 2005-2013 Various Mono authors

Build started 7/2/2020 1:29:52 PM.
__________________________________________________
Project "/home/tcal/r2/renode/lib/cctask/CCTask.sln" (default target(s)):
        Target ValidateSolutionConfiguration:
/home/tcal/r2/renode/lib/cctask/CCTask.sln: error : Invalid solution configuration and platform: "Release|FOOBAR".
        Task "Error" execution -- FAILED
        Done building target "ValidateSolutionConfiguration" in project "/home/tcal/r2/renode/lib/cctask/CCTask.sln".-- FAILED
Done building project "/home/tcal/r2/renode/lib/cctask/CCTask.sln".-- FAILED

Build FAILED.
Errors:

/home/tcal/r2/renode/lib/cctask/CCTask.sln (default targets) ->
(ValidateSolutionConfiguration target) ->

        /home/tcal/r2/renode/lib/cctask/CCTask.sln: error : Invalid solution configuration and platform: "Release|FOOBAR".

         0 Warning(s)
         1 Error(s)

Time Elapsed 00:00:00.1329520
```

PiotrZierhoffer, 2020-07-03:
As the PLATFORM env can be legitimately used in the `build.sh` context, I don't know if we should prevent setting it - but the error should not be silent for sure.

mateusz-holenko, 2021-05-05:
This is fixed in c1f83f13c00204d099bef51195f6f96c524168be.

closed 2021-05-05 by mateusz-holenko