renode/renode #107: Printfs now showing

mansnils, 2020-07-06:
I noticed that printfs are not coming through, although context-ReportError seems to work.

For example (after cloning tensorflow), put this in Prepare() in tensorflow/lite/micro/kernels/cmsis-nn/conv.cc
  printf("printf: this is not seen\n");
  context->ReportError(context, "ReportError: This is printed");
And run
make  -f tensorflow/lite/micro/tools/make/Makefile TAGS=cmsis-nn TARGET=stm32f4 test_kernel_conv_test

As long as ReportError() can be used for printing, this is not a big deal. I was just wondering why printfs are not printed. Is it some configuration issue?

PiotrZierhoffer, 2020-07-22:
It seems that `printf` uses a semihosting operation we do not support. We'll see how easy it is to add it.

joge1702, 2025-01-31:
@PiotrZierhoffer are there any ongoing efforts or plans to support semihosting in renode?