renode/renode #108: Running Robot tests should not require cloning the repository

franc0is, 2020-07-08:
At the moment, running tests with the Robot Framework & Renode requires:
1. A recent renode build
2. A clone of the repository and its submodules

(2) is required so the `test.sh` script as well as some of the python libraries can be used. Ideally, all of this should be contained within the renode install.

I suggest a Monitor command to run a `.robot` file, and/or a command line flag on Renode.exe.

mateusz-holenko, 2020-07-22:
Hi @franc0is!

Thanks for reporting the issue.

I verified this and confirm there are some problems* with robot tests in the latest Renode release.
In fact, tests themselves run fine, but the package lacks a file needed to display progress on the console - that's why it looks like nothing happens. But if you wait long enough for all the tests to finish, you should see log, report and xml files with the summary and results.

The package generation is now fixed and we are preparing a new release - expect it the next week.

mateusz-holenko, 2020-08-10:
[Renode 1.10](https://github.com/renode/renode/releases/tag/v1.10.1) is released - running tests should now work without the need of cloning the sources.

Please bare in mind that you still need to install some additional python libraries using:
`pip3 install -r /opt/renode/tests/requirements.txt`

closed 2020-08-10 by mateusz-holenko