renode/renode #110: STM32F4 freezing at I2C registration

psychodelly, 2020-07-10:
I am trying to add a custom I2C peripheral. It needs to change its I2C slave address upon changing the corresponding register value and register/deregister itself depending on a GPIO being set or unset. When the simulation is paused or not started yet, it works just fine. During runtime, however, the simulation freezes when executing the Register() method and the task has to be killed; Unregister() works. 

The peripheral starts registered at a GPIO Port with the I2C bus to connect to as a parameter:
```
LIDAR1: VL53L1x @ gpioPortE
    stm32f4_i2c: i2c1

gpioPortE:
    8 -> LIDAR1@0
```

```csharp
public VL53L1x(STM32F4_I2C stm32f4_i2c) 
{
    ShutdownPin = new GPIO();
    Bus = stm32f4_i2c;
    hwStandby = true;
    Reset();
}

public void OnGPIO(int number, bool value)
{
    if(value == false && hwStandby == false)
    {
        this.Log(LogLevel.Debug, "Shutdown requested for lidar");
        Bus.Unregister(this);
        hwStandby = true;
    }
    else if(value == true && hwStandby == true)
    {
	hwStandby = false;
        this.Log(LogLevel.Debug, "Lidar booting");
        reg = new NumberRegistrationPoint<int>(i2cSlaveDeviceAddress);
        Bus.Register(this, reg);
    }
}
```

mateusz-holenko, 2021-05-12:
The Register/Unregister API is dedicated for platform initialization and has not been intended for dynamic changes during the simulation.

An option is to register the device on both addresses directly in repl (if you know the expected addresses upfront, ofc):

```
LIDAR1: VL53L1x @ { i2c1 5; i2c1 7 }
```

Would it be enough for your case?

PiotrZierhoffer, 2021-09-21:
Let's close this for now, please reopen if you have any other comments

closed 2021-09-21 by PiotrZierhoffer