renode/renode #113: Renode v1.10 crashes when running

lbmeng, 2020-07-29:
Renode, version 1.10.0.27546 (52e0887a-202007271503)

```
(monitor) include @scripts/single-node/icicle-kit.resc 
(machine-0) start
```

several seconds later, all 4 windows (2 uart windows, monitor window, main window) crashed, and the program exists silently.

I am using the Windows build downloaded from https://renode.io/.
My Windows version: Win 10 version 1909 Build 18363.959

PiotrZierhoffer, 2020-07-29:
Thanks for the report. We can confirm it on Windows, and will investigate it

PiotrZierhoffer, 2020-07-29:
@lbmeng can you please open Windows "Event Viewer" just after the crash?
Go to `Windows Logs -> Applications`, find the last error with `.NET Runtime` as `Source` and paste the `Exception info` from the text box?
[image]

lbmeng, 2020-07-29:
Here it is:

```
Application: Renode.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.InvalidOperationException
   at Antmicro.Renode.Peripherals.CPU.TranslationCPU.TouchHostBlock(UInt64)
   at Antmicro.Renode.Peripherals.CPU.TranslationCPU+SimpleMemoryManager.Free(IntPtr)
   at Antmicro.Renode.Peripherals.CPU.TranslationCPU.ExecuteInstructions(UInt64, UInt64 ByRef)
   at Antmicro.Renode.Peripherals.CPU.TranslationCPU.CpuThreadBodyInner(Boolean)
   at Antmicro.Renode.Peripherals.CPU.TranslationCPU.CpuThreadBody()
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object)
   at System.Threading.ThreadHelper.ThreadStart()

```

lbmeng, 2020-07-29:
Is this caused by .NET version?

From https://github.com/renode/renode/#installation, it says:

```
Renode requires Mono >= 5.20 (Linux, macOS) or .NET >= 4.7 (Windows).
Windows 10 ships with .NET by default, so no action is required there.
```

So Windows 10 shipped .NET version does not satisfy the >= 4.7 requirement?

PiotrZierhoffer, 2020-07-29:
We believe it's not related, as it happens on our setups as well.

PiotrZierhoffer, 2020-07-30:
This is fixed on master. Thank you for your report. We will prepare a hotfix release to cover this.

closed 2020-07-30 by PiotrZierhoffer

lbmeng, 2020-07-31:
> This is fixed on master. Thank you for your report. We will prepare a hotfix release to cover this.

Thank you. I confirmed that **Renode, version 1.10.1.13150 (b7e0773f-202007300705)** fixed this issue.