renode/renode #114: I2C communication not working on STM32F4

psychodelly, 2020-07-29:
When using standard STM32F4 HAL functions (Mem_Read and Mem_Write, also tried Master_Transmit, didn't test Master_Receive, but should be similar looking at the HAL code) with the Renode STM32F4_I2C controller, the program is stuck in an infinite loop or the communication runs into timeout. 
I've debugged it with GDB and in both cases it is waiting for the BTF flag to be set.

The Mem_Write issue can be fixed by setting the BTF value to true in the DataWrite function in case "State.AwaitingData":

[image]

The Mem_Read issue only occurs when trying to read two bytes. It happens due to its different handling in the HAL code compared to reading a single or more than two bytes: When in 2-byte mode, the POS bit of the Control1 register is set after sending slave and memory address to indicate the first byte being read into the data register and the second byte into the shift register. Before starting to read the first byte from the data register the BTF flag is checked, resulting in a deadlock or timeout. 
This can be fixed by adding a callback to the POS bit, setting the BTF value to true and changing a flag to indicate the communication being in 2-byte mode: 

[image]
[image]

This flag needs to be checked in the StartWrite function to avoid overwriting the BTF value before starting to read out the registers:

[image]

psychodelly, 2020-08-06:
Additionally, I've noticed that I2C writing to Renode sensor implementations does not work (tested with BMA180). The sensor always changes its state to "SendingData" regardless of a Write or Read operation being requested. 
In the sensor code (Write(byte[] data) function) the first byte of a Write operation is incorrectly recognized as the device state; this byte will always be just the sensor's I2C slave address shifted left by 1 bit, since even in case of a Read operation, a normal Write operation is executed first. This Write operation preceding a Read operation can be detected via the data length, as it only contains device address and register address:

[image]

This also leads to the States enum being wrong: ReceivingData should be the device slave address shifted left by 1 bit and SendingData should be ReceivingData increased by 1, due to the LSB being 1 for a Read and 0 for a Write operation.

timmyhoa, 2023-10-21:
Heya, I'm running into the same problem. Has you fixed it already?