renode/renode #117: Running icicle-kit demo the Linux kernel doest not boot to shell with Renode v1.10.1

lbmeng, 2020-07-31:
Renode, version 1.10.1.13150 (b7e0773f-202007300705)

Executing:
```
(monitor) include @scripts/single-node/icicle-kit.resc 
(machine-0) s
Starting emulation...
```

The Linux kernel does not boot to shell in the **machine-0:sysbus.mmuart1** Window. It hangs at:
```
[    1.183799] This architecture does not have kernel memory protection.
[    1.183971] Run /init as init process
[    1.184019]   with arguments:
[    1.184063]     /init
[    1.184176]     earlyprintk
[    1.184242]   with environment:
[    1.184292]     HOME=/
[    1.184324]     TERM=linux
[    1.190351] mmc0: unrecognised CSD structure version 0
[    1.190460] mmc0: error -22 whilst initialising MMC card
[    1.231849] mmc0: unrecognised CSD structure version 0
[    1.231959] mmc0: error -22 whilst initialising MMC card
[    1.274891] mmc0: unrecognised CSD structure version 0
[    1.275008] mmc0: error -22 whilst initialising MMC card
[    1.317787] mmc0: unrecognised CSD structure version 0
[    1.317896] mmc0: error -22 whilst initialising MMC card
Starting syslogd: OK
Starting klogd: OK
Running sysctl: OK
Starting mdev... OK
```

No any output is seen after this point.

lbmeng, 2020-07-31:
I tried the docker image **antmicro/renode** and the same commands can boot to Linux shell.

mateusz-holenko, 2020-07-31:
Hi @lbmeng, thanks for reporting the issue!

Since this looks like a non-deterministic problem on a multi-core system, could you change the emulation settings to execute CPUs in serial using the following command in Monitor:

```
(monitor) i @scripts/single-node/icicle-kit.resc
(machine-0) emulation SetGlobalSerialExecution true
(machine-0) s
```

This increases the determinism of execution, so it should be easier to debug the underlying problem. Please bare in mind though, that this might also increase the boot time.

If that doesn't help, could you also change the quantum parameter for the simulation with:

```
(monitor) i @scripts/single-node/icicle-kit.resc
(machine-0) emulation SetGlobalSerialExecution true
(machine-0) emulation SetGlobalQuantum "0.001"
(machine-0) s
```

and report if this helps?

lbmeng, 2020-08-03:
> ```
> (monitor) i @scripts/single-node/icicle-kit.resc
> (machine-0) emulation SetGlobalSerialExecution true
> (machine-0) s
> ```

This does not help.

> If that doesn't help, could you also change the quantum parameter for the simulation with:
> 
> ```
> (monitor) i @scripts/single-node/icicle-kit.resc
> (machine-0) emulation SetGlobalSerialExecution true
> (machine-0) emulation SetGlobalQuantum "0.001"
> (machine-0) s
> ```
> 
> and report if this helps?

This helps. With this changes, the Linux kernel boots to shell.

mateusz-holenko, 2020-08-04:
Thanks for the report.

I'm now able to reproduce the problem locally and will investigate this.

wojteklewandowski-pl, 2020-09-24:
Hi Mateusz, is there something new in this topic? I have  this issue from time to time, but generally it hangs for a long time after Startng mdev .. ok.

PiotrZierhoffer, 2020-10-01:
Hi @wojteklewandowski-pl 

While we do not have it put in a release yet, the Icicle script on current `master` should boot 100% times.
It actually uses the settings suggested by @mateusz-holenko. We still try to find the underlying cause, but this is not a bad workaround, so we decided to publish it.

wojteklewandowski-pl, 2020-10-02:
Hi Piotr,
Thank you for your reply. Yes, it is not a bad workaround and it generally works and helps. I have noticed that when I recompiled files for Icicle because I have added many more things that I want to test, it sometimes hangs even with a workaround. 
I am waiting a while after: 
Starting mdev... OK
If this:
Populating /dev using udev: starting version 3.2.9
won't show up, the Linux won't boot and it hangs. If its show up Linux will boot.

BTW. I don't want to create another issue for that question (if I have to let me know). Does the renode support Linux dynamic device management with devtmpfs + mdev and devtmpfs + eudev? I recon yes, but want to make sure because I am currently struggling with /dev/input events and /dev/i2c.

PiotrZierhoffer, 2020-10-06:
Hi Wojtek

If you have a setup (or just relevant config files) that fails in a reproducible way, we'd be happy to take a look.

Regarding your question - yes, Renode is in general software agnostic, and as we aim to support whole SoCs, the way devices are handled by the OS should not be an issue. Of course we might miss some additional models or registers relevant in the process, but conceptually we do not "support" any specific way of device management, we expect all of them to work.

wojteklewandowski-pl, 2020-10-19:
Hi Piotr, 
In this [case](https://github.com/renode/renode/issues/136#issuecomment-711987678) I have posted my Linux files for renode. It sometimes hangs after Starting mdev... Maybe you can find something. 
Regards
Wojtek

Shivaraj199, 2023-03-17:
Hello I am facing this same issue the tried 

(monitor) i @scripts/single-node/icicle-kit.resc
(machine-0) emulation SetGlobalSerialExecution true
(machine-0) emulation SetGlobalQuantum "0.001"
(machine-0) s

but the boot gets stuck at mdev

AditCha, 2024-06-21:
Hi,

The issue still persists. I have tried running the provided demo script with various values for GlobalQuantum, as well as without them on Docker instances for multiple versions 10.1, 13.3, and 15. But to no success. 
 Is there any other suggested workaround?

martinjonsson01, 2026-09-02:
Hi,

We have recently implemented an atomics overhaul (see commit 26b4b1c) that fixed the correctness of LR/SC operations. I've tested the Icicle-Kit demo before and after this commit and have verified that it fixes the issue.

The workaround has now been removed from the `.resc` in commit 493f27a.

Please try it out and if you encounter problems please open a new issue!

closed 2026-09-02 by martinjonsson01