renode/renode #118: $bin does not work as expected for icicle-kit demo

lbmeng, 2020-07-31:
Renode, version 1.10.1.13150 (b7e0773f-202007300705)

After specifying the $bin:
```
(monitor) $bin=@images/hss.elf
(monitor) i @scripts/single-node/icicle-kit.resc
```

The script still tries to download images as specified in **polarfire-soc.resc**
```
21:28:18.1920 [INFO] Downloading https://dl.antmicro.com/projects/renode/pse-freertos-lwip.elf-s_1413416-f348e11b20bd1a8cb24c39e888cf6430e4ee4035.
21:28:22.0601 [INFO] Downloading: https://dl.antmicro.com/projects/renode/pse-freertos-lwip.elf-s_1413416-f348e11b20bd1a8cb24c39e888cf6430e4ee4035
    Progress: 6% (96KiB/1.35MiB)
    Speed: 51.37KB/s.
```

It looks to me $bin does not work as expected.

PiotrZierhoffer, 2020-07-31:
It does download the file, but in the end loads the one you specified. It's a peculiarity of the Monitor. After the elf is loaded, the log should mention the actual path (I am unable to verify this at the moment though).

lbmeng, 2020-07-31:
It looks **?=** usage is undocumented. The renode documentation only mentions **=**.

So my understanding is that if $bin=foo and $bin?=bar, $bin is foo (similar to Makefile variable).

> It's a peculiarity of the Monitor grammar.
So how to get rid of this annoying download every time if I don't have any cached files locally?