renode/renode #121: an error executing command 'emulation AddSyncDomain'

policeman0077, 2020-08-28:
os: ubuntu 20.04lts
renode: 1.10.1.26379 (b7e0773f-202007301407)
the example i tried to run is contiki-ng's rpl-udp
after startup the monitor shows:
`There was an error executing command 'emulation AddSyncDomain'`
`emulation does not provide a field, method or property AddSyncDomain.`

[image]

PiotrZierhoffer, 2020-08-28:
This is due to the script using Renode commands from an older release.

We do have a demo using this example here: https://github.com/renode/renode/blob/master/scripts/multi-node/cc2538/rpl-udp.resc

You can alter it to use the binaries you create. But we will see into updating the relevant scripts in the contiki-ng repo as well.

policeman0077, 2020-08-28:
@PiotrZierhoffer thank you Piotr. I modified the files accordingly and now the rpl-udp eaxmple seems working.
I enclosed the modified files here:

cc2538-node.resc
```
#:name: CC2538
#:description: This script runs Contiki-NG hello-world example on CC2538.

#$name?="CC2538"
#$bin?=@https://dl.antmicro.com/projects/renode/cc2538-contiki_hello_world.elf-s_242120-08fc83d11f790ccc1aa46abfdfc9c2e1a94baed2
#$bin=@hello-world.cc2538dk

mach create $name
using sysbus
machine LoadPlatformDescription @platforms/cpus/cc2538.repl

machine PyDevFromFile @scripts/pydev/rolling-bit.py 0x400D2004 0x4 True "sysctrl"

$id = `next_value 1`

macro reset
"""
    #set node address. 0x00 0x12 0x4B is TI OUI
    sysbus WriteDoubleWord 0x00280028 $id
    sysbus WriteDoubleWord 0x0028002C 0x00
    sysbus WriteDoubleWord 0x00280030 0xAB
    sysbus WriteDoubleWord 0x00280034 0x89
    sysbus WriteDoubleWord 0x00280038 0x00
    sysbus WriteDoubleWord 0x0028003C 0x4B
    sysbus WriteDoubleWord 0x00280040 0x12
    sysbus WriteDoubleWord 0x00280044 0x00

    sysbus LoadBinary @https://dl.antmicro.com/projects/renode/cc2538_rom_dump.bin-s_524288-0c196cdc21b5397f82e0ff42b206d1cc4b6d7522 0x0
    sysbus LoadELF $bin

    cpu VectorTableOffset `sysbus GetSymbolAddress "vectors"`
```
udp-rpl.resc
```
:name: cc2538 range medium 2 nodes

######################################################################
#
# Sample script creating 1 server node and 1 client node using a
# range-based medium function - transmission is possible if nodes are
# in range. Nodes are positioned in a 3D space (x,y,z),
# at a distance of 10 (units) from each other (on the X axis).
#
# Differences in SyncUnit values in machines represent the differences
# in their relative speed. This prevents the machines from
# synchronizing in 100%, which is crucial for Contiki network stack
# stability.
#
######################################################################

#path add $CWD
emulation CreateWirelessMedium "wireless"
#$domain = `emulation AddSyncDomain`
#wireless SetSyncDomainFromEmulation $domain
#wireless SetRangeWirelessFunction 11

# The radio is using a range-based medium (with the `Range` set to `11`)
# If not set, the default SimpleMediumFunction will be used (where range is not relevant)
wireless SetRangeWirelessFunction 11

######################### UDP SERVER - begin #########################
$bin=@udp-server.cc2538dk
$name="server"
i $ORIGIN/cc2538-node.resc
connector Connect radio wireless
wireless SetPosition radio 0 0 0
#machine SyncUnit 100
#showAnalyzer sysbus.uart0
mach clear
########################## UDP SERVER - end ##########################

######################### UDP CLIENT - begin #########################
$bin=@udp-client.cc2538dk
$name="client"
i $ORIGIN/cc2538-node.resc
connector Connect radio wireless
wireless SetPosition radio 10 0 0
#machine SyncUnit 110
#showAnalyzer sysbus.uart0
mach clear
########################## UDP CLIENT - end ##########################
```

policeman0077, 2020-08-28:
@PiotrZierhoffer by the way is there a good reference on how to write resc files and what does all those commands mean?  There is not a lot of stuff in https://renode.readthedocs.io/en/latest/introduction/using.html#resc-scripts . 
thank you.