renode/renode #122: uart transmitter is not enabled -- zephyr on stm32f3_disco

swappad, 2020-08-28:
I am attempting to simulate zephyr's "samples/helloworld" demo for stm32f3 discovery board.
I created a new platform similar to already existing platform descriptions of stm32f* and changed peripheral addresses accordingly. I am using the official svd for stm32f.

The demo does work, when I build for stm32f4_disco and use the supported platform from renode, but for the new stm32f3 platform I don't get any text printed on any uart bus:

```
22:46:10.8158 [INFO] sysbus: Loading segment of 31940 bytes length at 0x8000000.
22:46:10.8254 [INFO] sysbus: Loading segment of 380 bytes length at 0x8007CC4.
22:46:10.8255 [INFO] sysbus: Loading segment of 7696 bytes length at 0x20000180.
22:47:16.4064 [INFO] cpu: Guessing VectorTableOffset value to be 0x8000000.
22:47:16.4110 [INFO] cpu: Setting initial values: PC = 0x80028CD, SP = 0x20001450.
22:47:16.4138 [INFO] machine-0: Machine started.
22:47:16.4666 [WARNING] nvic: Unhandled read from offset 0xF34.
22:47:16.4667 [WARNING] nvic: Unhandled write to offset 0xF34, value 0x0.
22:47:16.4752 [WARNING] nvic: Unhandled read from offset 0xD14.
22:47:16.4753 [WARNING] nvic: Unhandled write to offset 0xD14, value 0x200.
22:47:16.4756 [WARNING] nvic: Unhandled read from offset 0xD14.
22:47:16.4757 [WARNING] nvic: Unhandled write to offset 0xD14, value 0x10.
22:47:16.4758 [WARNING] nvic: Unhandled value written to System Control Register: 0x10.
22:47:16.4759 [WARNING] nvic: Unhandled write to offset 0xD28, value 0xFFFFFFFF.
22:47:16.4760 [WARNING] nvic: Unhandled write to offset 0xD2C, value 0xFFFFFFFF.
22:47:16.6041 [WARNING] uart2: Unhandled write to offset 0x0. Unhandled bits: [2] when writing value 0xCC. Tags: NF (0x1).                                                                                                 
22:47:16.6049 [WARNING] uart2: Trying to transmit a character, but the transmitter is not enabled. dropping.                                                             
22:47:16.6058 [WARNING] uart2: Unhandled write to offset 0xC. Unhandled bits: [0-1] when writing value 0x23. Tags: SBK (0x1), RWU (0x1).                                                                                                
22:47:16.6059 [WARNING] uart2: Unhandled write to offset 0x0. Unhandled bits: [0] when writing value 0xC1. Tags: PE (0x1).                                                                                                 
22:47:22.4856 [WARNING] uart2: Unhandled read from offset 0x1C. (91608)
```

This is the platform description:
```
fsmcBank1: Memory.MappedMemory @ sysbus 0x60000000
    size: 0x10000000

ccm: Memory.MappedMemory @ sysbus 0x10000000 
    size: 0x00002000

sram: Memory.MappedMemory @ sysbus 0x20000000
    size: 0x0000a000

flash: Memory.MappedMemory @ sysbus 0x08000000
    size: 0x00040000

nvic: IRQControllers.NVIC @ sysbus 0xE000E000
    priorityMask: 0xF0
    systickFrequency: 72000000
    IRQ -> cpu@0

cpu: CPU.CortexM @ sysbus
    cpuType: "cortex-m4"
    nvic: nvic

i2c1: I2C.STM32F4_I2C @ sysbus 0x40005400
    EventInterrupt -> nvic@31
    ErrorInterrupt -> nvic@32

i2c2: I2C.STM32F4_I2C @ sysbus 0x40005800
    EventInterrupt -> nvic@33
    ErrorInterrupt -> nvic@34

uart1: UART.STM32_UART @ sysbus <0x40013800, +0x100>
    -> nvic@37
    frequency: 115200

uart2: UART.STM32_UART @ sysbus <0x40004400, +0x100>
    -> nvic@38
    frequency: 115200

uart3: UART.STM32_UART @ sysbus <0x40004800, +0x100>
    -> nvic@39

rcc: Python.PythonPeripheral @ sysbus 0x40021000
    size: 0x400
    initable: true
    filename: "scripts/pydev/flipflop.py"

dma1: DMA.STM32DMA @ sysbus 0x40020000
    [0-6] -> nvic@[11-17]

dma2: DMA.STM32DMA @ sysbus 0x40020400
    [0-4] -> nvic@[56-60]

exti: IRQControllers.EXTI @ sysbus 0x40010400
    [0-6] -> nvic@[6-10, 23, 40]

gpioPortA: GPIOPort.STM32F4GPIOPort @ sysbus <0x48000000, +0x400>
    [0-15] -> exti@[0-15]

gpioPortB: GPIOPort.STM32F4GPIOPort @ sysbus <0x48000400, +0x400>
    [0-15] -> exti@[0-15]

gpioPortC: GPIOPort.STM32F4GPIOPort @ sysbus <0x48000800, +0x400>
    [0-15] -> exti@[0-15]

gpioPortD: GPIOPort.STM32F4GPIOPort @ sysbus <0x48000c00, +0x400>
    [0-15] -> exti@[0-15]

gpioPortE: GPIOPort.STM32F4GPIOPort @ sysbus <0x48001000, +0x400>
    [0-15] -> exti@[0-15]

gpioPortF: GPIOPort.STM32F4GPIOPort @ sysbus <0x48001400, +0x400>
    [0-15] -> exti@[0-15]

sysbus:
    init:
        ApplySVD @./STM32F3_svd_V1.2/STM32F303.svd
        Tag <0x40021000, 0x40021003> "RCC_CR" 0x03037683
```

PiotrZierhoffer, 2021-05-11:
Looking at the reference manuals, it looks like you should use another UART model. STM32F3 is in this regard compatible with STM32F7, so change your repl to:

```
uart1: UART.STM32F7_USART @ ...
```

PiotrZierhoffer, 2021-05-11:
I close this issue now, but if you get to check it and see it does not work - please open a new one.

closed 2021-05-11 by PiotrZierhoffer