renode/renode #123: icicle-kit demo - mmc timeout

wojteklewandowski-pl, 2020-09-02:
Hello,
I have successfully run the icycle-kit demo you have provided, but have problems with files built by myself. I have been building files using [this repo](https://github.com/polarfire-soc/polarfire-soc-buildroot-sdk). The problem is with emmc.img. My emmc.img is exactly the same size as yours, but when I use all my files, except emmc.img (use emmc.img from your demo) the linux boot, however when I use all my files including my emmc image I have such a problem:

**mmuart1**

```
RAM:  1 GiB
MMC:   sdhc@20008000: 0
In:    serial@20100000
Out:   serial@20100000
Err:   serial@20100000
Net:   
Warning: ethernet@20112000 using MAC address from ROM
eth0: ethernet@20112000
Hit any key to stop autoboot:  0 
sdhci_transfer_data: Transfer data timeout
sdhci_send_command: Timeout for status update!
sdhci_send_command: Timeout for status update!
sdhci_send_command: Timeout for status update!
## Warning: Input data exceeds 1048576 bytes - truncated
## Info: input data size = 1048578 = 0x100002
Running boot cmd...
## Error: "boot3" not defined
RISC-V # 
```

**renode output during this timeouts**

```
1:24:23.7119 [WARNING] mmc: Unhandled write to offset 0x22C. Unhandled bits: [17-19] when writing value 0xE0002.
11:24:23.7121 [WARNING] mmc: Unhandled write to offset 0x204. Unhandled bits: [12-14] when writing value 0x17008.
11:24:23.7122 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [1, 4] when writing value 0x37020012. Tags: Block Count Enable (BCE) (0x1), Data Transfer Direction Select (DTDS) (0x1).
11:24:23.7124 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [19-21] when writing value 0x333A0000. Tags: Command CRC Check Enable (CRCCE) (0x1), Command Index Check Enable (CICE) (0x1), Data Present Select (DPS) (0x1).
11:26:12.9430 [WARNING] mmc: Unhandled write to offset 0x230. Unhandled bits: [9-13, 26, 28-31] when writing value 0xFFFFFFFF.
11:26:12.9430 [WARNING] mmc: Unhandled write to offset 0x22C. Unhandled bits: [25] when writing value 0x2000002. Tags: Software Reset For CMD Line (SRCMD) (0x1).
11:26:12.9431 [WARNING] mmc: Unhandled write to offset 0x22C. Unhandled bits: [26] when writing value 0x4000002. Tags: Software Reset For DAT Line (SRDAT) (0x1).
11:26:12.9435 [WARNING] mmc: Unhandled write to offset 0x230. Unhandled bits: [9-13, 26, 28-31] when writing value 0xFFFFFFFF.
11:26:12.9435 [WARNING] mmc: Unhandled write to offset 0x22C. Unhandled bits: [17-19] when writing value 0xE0002.
11:26:12.9435 [WARNING] mmc: Unhandled write to offset 0x204. Unhandled bits: [12-14] when writing value 0x17008.
11:26:12.9437 [WARNING] mmc.sdCard: Unsupported command: 51. Ignoring it
11:26:12.9438 [WARNING] mmc: Expected a response of length 32 bits (excluding control bits and CRC), but 0 received
11:26:12.9439 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [1, 4] when writing value 0x33020012. Tags: Block Count Enable (BCE) (0x1), Data Transfer Direction Select (DTDS) (0x1).
11:26:12.9439 [WARNING] mmc.sdCard: Unsupported command: 51. Ignoring it
11:26:12.9439 [WARNING] mmc: Expected a response of length 32 bits (excluding control bits and CRC), but 0 received
11:26:12.9439 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [19-21] when writing value 0x333A0000. Tags: Command CRC Check Enable (CRCCE) (0x1), Command Index Check Enable (CICE) (0x1), Data Present Select (DPS) (0x1).
11:26:22.1236 [WARNING] mmc: Unhandled write to offset 0x230. Unhandled bits: [9-13, 26, 28-31] when writing value 0xFFFFFFFF.
11:26:22.1237 [WARNING] mmc: Unhandled write to offset 0x22C. Unhandled bits: [17-19] when writing value 0xE0002.
11:26:22.1237 [WARNING] mmc: Unhandled write to offset 0x204. Unhandled bits: [12-14] when writing value 0x17008.
11:26:22.1238 [WARNING] mmc.sdCard: Unsupported command: 51. Ignoring it
11:26:22.1238 [WARNING] mmc: Expected a response of length 32 bits (excluding control bits and CRC), but 0 received
11:26:22.1238 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [1, 4] when writing value 0x33020012. Tags: Block Count Enable (BCE) (0x1), Data Transfer Direction Select (DTDS) (0x1).
11:26:22.1239 [WARNING] mmc.sdCard: Unsupported command: 51. Ignoring it
11:26:22.1239 [WARNING] mmc: Expected a response of length 32 bits (excluding control bits and CRC), but 0 received
11:26:22.1239 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [19-21] when writing value 0x333A0000. Tags: Command CRC Check Enable (CRCCE) (0x1), Command Index Check Enable (CICE) (0x1), Data Present Select (DPS) (0x1).
11:26:31.3120 [WARNING] mmc: Unhandled write to offset 0x230. Unhandled bits: [9-13, 26, 28-31] when writing value 0xFFFFFFFF.
11:26:31.3120 [WARNING] mmc: Unhandled write to offset 0x22C. Unhandled bits: [17-19] when writing value 0xE0002.
11:26:31.3120 [WARNING] mmc: Unhandled write to offset 0x204. Unhandled bits: [12-14] when writing value 0x17008.
11:26:31.3121 [WARNING] mmc.sdCard: Unsupported command: 51. Ignoring it
11:26:31.3121 [WARNING] mmc: Expected a response of length 32 bits (excluding control bits and CRC), but 0 received
11:26:31.3121 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [1, 4] when writing value 0x33020012. Tags: Block Count Enable (BCE) (0x1), Data Transfer Direction Select (DTDS) (0x1).
11:26:31.3122 [WARNING] mmc.sdCard: Unsupported command: 51. Ignoring it
11:26:31.3123 [WARNING] mmc: Expected a response of length 32 bits (excluding control bits and CRC), but 0 received
11:26:31.3123 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [19-21] when writing value 0x333A0000. Tags: Command CRC Check Enable (CRCCE) (0x1), Command Index Check Enable (CICE) (0x1), Data Present Select (DPS) (0x1).
```

Can you help me with that?
One general question about the simulation flow. Accordingly to the resc file we have to provide a fitImage, kernel, hss, u-boot and emmc image. From what I've checked the fit image is used among others to create the emmc image, so what is a reason for providing the fit image separately? If I use my files, and your emmc which fit image is used, this from emmc image, or this provided explicitly?
Regards,
Wojtek

wojteklewandowski-pl, 2020-09-04:
Hi, I narrowed down the problem. It is related only to the uboot. Your example has different bootcmd (`bootcmd=echo 'Running Linux...'; bootm ${ramdisk_addr_r}`) than the bootcmd in the polarfire-buildroot-sdk (`bootcmd=setenv fileaddr 0x90000000; load mmc 0:2 ${fileaddr} uEnv_s-mode.txt; env import -t ${fileaddr} ${filesize}; echo 'Running boot cmd...'; run boot3`) 
Whith yours bootcmd I am able to boot. It looks that there is something wrong with communication to mmc in u-boot. I don't know why yet, but even mmcinfo in u-boot gives a timeout.
mmuart1
```
RISC-V # mmcinfo 
sdhci_transfer_data: Transfer data timeout
sdhci_send_command: Timeout for status update!
sdhci_send_command: Timeout for status update!
sdhci_send_command: Timeout for status update!
```
the terminal where renode has been started:
 ```
 1 08:14:24.4582 [WARNING] mmc: Unhandled read from offset 0x18.
  2 08:14:24.4583 [WARNING] mmc: Unhandled write to offset 0x18, value 0x0.
  3 08:14:24.4583 [WARNING] mmc: Unhandled read from offset 0x228. (2)
  4 08:14:24.4584 [WARNING] mmc: Unhandled write to offset 0x228, value 0x0.
  5 08:14:24.4785 [WARNING] mmc: Unhandled read from offset 0x18.
  6 08:14:24.4786 [WARNING] mmc: Unhandled write to offset 0x18, value 0x0.
  7 08:14:24.4786 [WARNING] mmc: Unhandled read from offset 0x228. (2)
  8 08:14:24.4787 [WARNING] mmc: Unhandled write to offset 0x228, value 0x0.
  9 08:14:24.4787 [WARNING] mmc: Unhandled read from offset 0x18.
 10 08:14:24.4787 [WARNING] mmc: Unhandled write to offset 0x18, value 0x0.
 11 08:14:24.4787 [WARNING] mmc: Unhandled write to offset 0x22C. Unhandled bits: [0, 15] when writing value 0x8001.
 12 08:14:24.4788 [WARNING] mmc: Unhandled write to offset 0x22C. Unhandled bits: [2] when writing value 0x6.
 13 08:14:24.4788 [WARNING] mmc: Unhandled read from offset 0x228. (2)
 14 08:14:24.4788 [WARNING] mmc: Unhandled write to offset 0x228, value 0x0.
 15 08:14:24.5196 [WARNING] mmc: Unhandled write to offset 0x230. Unhandled bits: [9-13, 26, 28-31] when writing value 0xFFFFFFFF. (3)
 16 08:14:24.5197 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [19-20] when writing value 0x81A0000. Tags: Command CRC Check Enable (CRCCE) (0x1), Command Index Chec    k Enable (CICE) (0x1).
 17 08:14:24.5297 [WARNING] mmc: Unhandled write to offset 0x230. Unhandled bits: [9-13, 26, 28-31] when writing value 0xFFFFFFFF. (2)
 18 08:14:24.5299 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [19-20] when writing value 0x371A0000. Tags: Command CRC Check Enable (CRCCE) (0x1), Command Index Che    ck Enable (CICE) (0x1).
 19 08:14:24.5515 [WARNING] mmc: Unhandled write to offset 0x230. Unhandled bits: [9-13, 26, 28-31] when writing value 0xFFFFFFFF. (4)
 20 08:14:24.5516 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [19] when writing value 0x2090000. Tags: Command CRC Check Enable (CRCCE) (0x1).
 21 08:14:24.5615 [WARNING] mmc: Unhandled write to offset 0x230. Unhandled bits: [9-13, 26, 28-31] when writing value 0xFFFFFFFF. (2)
 22 08:14:24.5616 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [19-20] when writing value 0x31A0000. Tags: Command CRC Check Enable (CRCCE) (0x1), Command Index Chec    k Enable (CICE) (0x1).
 23 08:14:24.5720 [WARNING] mmc: Unhandled write to offset 0x230. Unhandled bits: [9-13, 26, 28-31] when writing value 0xFFFFFFFF. (2)
 24 08:14:24.5721 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [19] when writing value 0x9090000. Tags: Command CRC Check Enable (CRCCE) (0x1).
 25 08:14:24.5822 [WARNING] mmc: Unhandled write to offset 0x230. Unhandled bits: [9-13, 26, 28-31] when writing value 0xFFFFFFFF. (2)
 26 08:14:24.5823 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [19-20] when writing value 0x71A0000. Tags: Command CRC Check Enable (CRCCE) (0x1), Command Index Chec    k Enable (CICE) (0x1).
 27 08:14:24.5932 [WARNING] mmc: Unhandled write to offset 0x230. Unhandled bits: [9-13, 26, 28-31] when writing value 0xFFFFFFFF. (2)
 28 08:14:24.5933 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [19-20] when writing value 0x371A0000. Tags: Command CRC Check Enable (CRCCE) (0x1), Command Index Che    ck Enable (CICE) (0x1).
 29 08:14:24.6042 [WARNING] mmc: Unhandled write to offset 0x230. Unhandled bits: [9-13, 26, 28-31] when writing value 0xFFFFFFFF. (2)
 30 08:14:24.6042 [WARNING] mmc: Unhandled write to offset 0x22C. Unhandled bits: [17-19] when writing value 0xE0002.
 31 08:14:24.6043 [WARNING] mmc: Unhandled write to offset 0x204. Unhandled bits: [12-14] when writing value 0x17008.
 32 08:14:24.6044 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [1, 4] when writing value 0x37020012. Tags: Block Count Enable (BCE) (0x1), Data Transfer Direction Se    lect (DTDS) (0x1).
 33 08:14:24.6045 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [19-21] when writing value 0x333A0000. Tags: Command CRC Check Enable (CRCCE) (0x1), Command Index Che    ck Enable (CICE) (0x1), Data Present Select (DPS) (0x1).
 34 08:16:11.9493 [WARNING] mmc: Unhandled write to offset 0x230. Unhandled bits: [9-13, 26, 28-31] when writing value 0xFFFFFFFF.
 35 08:16:11.9494 [WARNING] mmc: Unhandled write to offset 0x22C. Unhandled bits: [25] when writing value 0x2000002. Tags: Software Reset For CMD Line (SRCMD) (0x1).
[... cut ...]
 60 08:16:30.5990 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [1, 4] when writing value 0x33020012. Tags: Block Count Enable (BCE) (0x1), Data Transfer Direction Se    lect (DTDS) (0x1).
 61 08:16:30.5991 [WARNING] mmc.sdCard: Unsupported command: 51. Ignoring it
 62 08:16:30.5991 [WARNING] mmc: Expected a response of length 32 bits (excluding control bits and CRC), but 0 received
 63 08:16:30.5991 [WARNING] mmc: Unhandled write to offset 0x20C. Unhandled bits: [19-21] when writing value 0x333A0000. Tags: Command CRC Check Enable (CRCCE) (0x1), Command Index Che    ck Enable (CICE) (0x1), Data Present Select (DPS) (0x1).
```

Regards,
Wojtek

PiotrZierhoffer, 2020-09-04:
Hi Wojtek

Thanks for the report. This is correct, our booting flow is a little bit different to the official one, due to some issues with the MMC/SD controller.

Our current configuration has some changes that help us boot a bit faster and omitting a few rough edges we're still working on:
- we disable CONFIG_FIT_SIGNATURE and CONFIG_FIT_ENABLE_SHA256_SUPPORT in uBoot (performance reasons only. It also requires removing the hash algorithm from conf/osbi-fit-image.its )
- we change to credentials to root/root (as mentioned by the login prompt)
- we reduce hss and uBoot autoboot timeouts
- we configure uBoot with `CONFIG_BOOTCOMMAND="echo 'Running Linux...'; bootm ${ramdisk_addr_r}"`

This way, while HSS loads uBoot from SD, the kernel is preloaded to RAM by our script. We are now working on polishing the model of the controller, so that uBoot passes all the checks it performs.