renode/renode #124: segfault if quit is sent to monitor before disconnecting telnet from CreateServerSocketTerminal

y2kbugger, 2020-09-05:
```
=================================================================
        Native Crash Reporting
=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================
/home/y2k/devel/baremetal-riscv-renode/renode/renode: line 54: 19798 Aborted                 (core dumped) $LAUNCHER "${OUTPUT_PATH}/${TARGET}/Renode.exe" "${PARAMS[@]:-}"
```

This occurs if we issue `quit` to the monitor while we connected to 

```
emulation CreateServerSocketTerminal 1234 "externalUART" true
connector Connect uart externalUART
```

via telnet or netcat. emitConfig setting makes no difference.

If we issue quit _before connecting_ or _after disconnecting_ there is no problem.

two side issues, but I feel they are too obvious and as such are "can't fix". Let me know if I should create issues for these:
1. After a segfault, the terminal is left is a broken state. ie I can type but I cant see what I type, there is no echo. Readline and bash hotkeys still work though. Entering renode and exiting without a segfault fixes the terminal. [image]
2. There are certain times when I cannot kill renode via ctrl-c. This is actually most of the time. I actually noticed after a few months that I can get sometimes I can get one ctrl-c in that works and next time I need to `quit` via monitor. 
[image]
   It gets stuck after "Disposed". ctrl-c, ctrl-d have absolutely no effect at that point. only way to recover is to ctrl-z and kill

y2kbugger, 2020-09-05:
[image]

y2kbugger, 2020-09-05:
```
define mono_backtrace
 select-frame 0
 set $i = 0
 while ($i < $arg0)
   set $foo = (char*) mono_pmip ($pc)
   if ($foo)
     printf "#%d %p in %s\n", $i, $pc, $foo
   else
     frame
   end
   up-silently
   set $i = $i + 1
 end
end
```

PiotrZierhoffer, 2020-09-07:
The invalid state after a segfault is indeed a tough one - we'd prefer focusing on underlying issues to prevent segfaults.

The Ctrl+C issue is a problem we're aware of and we're looking for a proper, multi-os solution.

Regarding the main issue: I can't reproduce it locally and have never stumbled upon it, so I need to get more data.

- Can you please provide your ``mono --version`` output?
- Does it matter if you ``start`` the emulation?
- Do you run ``showAnalyzer`` on the UART as well? (it's not required at all, just looking for additional circumstances)
- Is there any data appearing on the UART (i.e. in the telnet/netcat window)? Do you input anything? Is the data read by software?

y2kbugger, 2020-09-07:
In addition to your question, I also reproduced whether or not I launch via `renode -P 45454 vexriscv.resc` or `renode -P 45454 vexriscv.resc`

> * Can you please provide your `mono --version` output?
```
$ mono --version
Mono JIT compiler version 6.10.0.104 (tarball Fri Jun 26 19:38:44 UTC 2020)
```
I'll give the renode version also, since I missed that:
```
Renode, version 1.10.1.36510 (7e978271-202009052017)
```
> * Does it matter if you `start` the emulation?

Unable to reproduce if I don't `start`

> * Do you run `showAnalyzer` on the UART as well? (it's not required at all, just looking for additional circumstances)

Not normally simultaneously:
But I just tested:
They both work simulateouly (both echoing correctly)
The segfault occurs if and only if CreateServerSocketTerminal is connnected via telnet (regardless of whether I call it in the `resc` file)

> * Is there any data visible on the UART (i.e. in the telnet/netcat window)? Do you input anything? Is the data read by software?
It would make sense if you could run `sysbus ShowPeripheralAccess uart` and paste the log

Doesn't matter whether I read or write anything. Right now I just echo back the the send chars with decoration e.g. send 'd' reply siimilar to "#d#", but the bug happens when I actually send anything or not. The only thing that mattters is whether I am currently connected. If I close telnet using the escape sequence `^]` even after using the uart for the "echo test" it exits fine when I issue quit in the monitor.

The only thing that is show is the one char read the the decorated echo (actually multiline so there are many chars)
```
renode vexriscv.resc
08:31:42.4845 [INFO] Loaded monitor commands from: /home/y2k/devel/baremetal-riscv-renode/renode/./scripts/monitor.py
08:31:42.8470 [INFO] Including script: /home/y2k/devel/baremetal-riscv-renode/x_uart/vexriscv.resc
08:31:42.8571 [INFO] System bus created.
08:31:43.4118 [INFO] sysbus: Loading segment of 506 bytes length at 0x10000.
08:31:43.4174 [INFO] sysbus: Loading segment of 20000 bytes length at 0x11000.
08:31:43.4327 [INFO] cpu: Setting PC value to 0x10080.
08:31:43.4577 [INFO] vexriscv-machine: GDB server with all CPUs started on port :3333
08:31:43.4641 [INFO] vexriscv-machine: Machine started.
08:31:43.5686 [INFO] sysbus: Logging already enabled: False.
08:31:50.0399 [INFO] uart: [cpu: 0x100D8] ReadByte from 0x0 (RxTx), returned 0x64.
08:31:50.0481 [INFO] uart: [cpu: 0x10108] WriteByte to 0x0 (RxTx), value 0x23. (3)
08:31:50.0482 [INFO] uart: [cpu: 0x10108] WriteByte to 0x0 (RxTx), value 0xA.
08:31:50.0483 [INFO] uart: [cpu: 0x10108] WriteByte to 0x0 (RxTx), value 0xD.
08:31:50.0483 [INFO] uart: [cpu: 0x10108] WriteByte to 0x0 (RxTx), value 0x23.
08:31:50.0484 [INFO] uart: [cpu: 0x10108] WriteByte to 0x0 (RxTx), value 0x64.
08:31:50.0485 [INFO] uart: [cpu: 0x10108] WriteByte to 0x0 (RxTx), value 0x23.
08:31:50.0485 [INFO] uart: [cpu: 0x10108] WriteByte to 0x0 (RxTx), value 0xA.
08:31:50.0486 [INFO] uart: [cpu: 0x10108] WriteByte to 0x0 (RxTx), value 0xD.
08:31:50.0488 [INFO] uart: [cpu: 0x10108] WriteByte to 0x0 (RxTx), value 0x23. (3)
08:31:50.0489 [INFO] uart: [cpu: 0x10108] WriteByte to 0x0 (RxTx), value 0xA.
08:31:50.0489 [INFO] uart: [cpu: 0x10108] WriteByte to 0x0 (RxTx), value 0xD.
08:31:50.0490 [INFO] uart: [cpu: 0x10108] WriteByte to 0x0 (RxTx), value 0xA. (2)
08:31:55.0429 [INFO] vexriscv-machine: Machine paused.
08:31:55.0661 [INFO] vexriscv-machine: Disposed.

=================================================================
        Native Crash Reporting
=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================
```
> * Does it matter if you start the GDB server or not?
reproduced with and without (presuming you mean `machine StartGdbServer 3333`
> * Have you tried running Renode interactively and reproducing this issue?

Just reproduced with and without.

> * Does it happen when using any other platform, not based on VexRiscv?

not sure, but my platform is pretty minimal

```
using sysbus
mach create $name
machine LoadPlatformDescription @vexriscv.repl
sysbus.cpu SetMachineIrqMask 0xFFFFFFFF

sysbus LoadELF @image

#machine StartGdbServer 3333

start

emulation CreateServerSocketTerminal 1234 "externalUART" true
connector Connect uart externalUART
```

I did some digging and it seems like there are a couple opportunities for things to get improperly disposed within `
renode/src/Infrastructure/src/Emulator/Main/Utilities/SocketServerProvider.cs` but nothing prodded at (dispose order, explicitly disposing `stream` ) prevented the segfault.