renode/renode #127: PythonPeripheral filename path doesn't respect system path

karlp, 2020-09-11:
If I use `path add blah` at the monitor, that works for my platform files, but I can't reference a PythonPeripheral filename like that.  It _must_ be path relative to the renode root, eg, `/opt/renode/`

ie

```
itm: Python.PythonPeripheral @ sysbus 0xe0000000
    size: 128 // 32 stimulus ports, 32bit wide.
    initable: false
    //script: "request.value = 1 if request.isRead"
    // note, must be in renode root dir, not in project :(
    filename: "renode.itm.py"
```

renode.itm.py must be in the root, not the project path, regardless of usage of @

majava3000, 2020-09-15:
I just noticed this as well. Absolute paths are accepted though. For example, I'm running the docker image (which has `/home/developer/` as the home directory. I placed my extra stuff under /home/developer/test and then used `filename: "/home/developer/test/foo.py"`

PiotrZierhoffer, 2020-10-06:
Hi @karlp 

This is indeed a limitation of file support in Renode. We do have a plan of changing the file access layer, but it's probably going to take some time.

For now the solution provided by @majava3000 is probably the easiest way to go