renode/renode #128: main console window scrolls oddly

karlp, 2020-09-11:
Sorry for the not awesome title, but not sure how else to describe it.  The console window where I started renode has the log, and normally it scrolls normally, but it sometimes jumps back to the first line of the window, and keeps replacing it's log entry there over and over again.  I suspect this is whenever it has a "repeated" line entry, in my case, the SPI write to a disconnected peripheral

text mode, note that _top_ line has a newer timestamp that the rest of the window.
```
13:52:12.6025 [WARNING] spi2: SPI transmission while no SPI peripheral is connected. (35920)
13:50:13.2876 [WARNING] sysbus: Write of value 0x2030405 to an unimplemented register Flash:PEKEYR (0x40023C0C) generated from SVD.
13:50:13.2877 [WARNING] sysbus: Read from an unimplemented register Flash:PECR (0x40023C04), returning a value from SVD: 0x7.
13:50:13.2877 [WARNING] sysbus: Write of value 0x7 to an unimplemented register Flash:PECR (0x40023C04) generated from SVD.
13:50:13.2877 [WARNING] sysbus: [cpu: 0x80085D2] WriteDoubleWord to non existing peripheral at 0x8080998, value 0x0.
13:50:13.2877 [WARNING] sysbus: Read from an unimplemented register Flash:PECR (0x40023C04), returning a value from SVD: 0x7.
13:50:13.2878 [WARNING] sysbus: Write of value 0x7 to an unimplemented register Flash:PECR (0x40023C04) generated from SVD.
13:50:13.2914 [WARNING] sysbus: [cpu: 0x80047C0] (tag: 'FLASH_SR') ReadDoubleWord from non existing peripheral at 0x40023C18, returning 0xE.
13:50:13.2929 [WARNING] sysbus: Write of value 0x89ABCDEF to an unimplemented register Flash:PEKEYR (0x40023C0C) generated from SVD.
13:50:13.2930 [WARNING] sysbus: Write of value 0x2030405 to an unimplemented register Flash:PEKEYR (0x40023C0C) generated from SVD.
13:50:13.2930 [WARNING] sysbus: Read from an unimplemented register Flash:PECR (0x40023C04), returning a value from SVD: 0x7.
13:50:13.2930 [WARNING] sysbus: Write of value 0x7 to an unimplemented register Flash:PECR (0x40023C04) generated from SVD.
13:50:13.2931 [WARNING] sysbus: [cpu: 0x80085D2] WriteDoubleWord to non existing peripheral at 0x808099C, value 0x0.
13:50:13.2931 [WARNING] sysbus: Read from an unimplemented register Flash:PECR (0x40023C04), returning a value from SVD: 0x7.
13:50:13.2931 [WARNING] sysbus: Write of value 0x7 to an unimplemented register Flash:PECR (0x40023C04) generated from SVD.
13:50:13.3033 [WARNING] sysbus: [cpu: 0x80047C0] (tag: 'FLASH_SR') ReadDoubleWord from non existing peripheral at 0x40023C18, returning 0xE.
13:50:13.3067 [WARNING] sysbus: Write of value 0x89ABCDEF to an unimplemented register Flash:PEKEYR (0x40023C0C) generated from SVD.
13:50:13.3068 [WARNING] sysbus: Write of value 0x2030405 to an unimplemented register Flash:PEKEYR (0x40023C0C) generated from SVD.
13:50:13.3068 [WARNING] sysbus: Read from an unimplemented register Flash:PECR (0x40023C04), returning a value from SVD: 0x7.
13:50:13.3068 [WARNING] sysbus: Write of value 0x7 to an unimplemented register Flash:PECR (0x40023C04) generated from SVD.
13:50:13.3069 [WARNING] sysbus: [cpu: 0x800482E] WriteByte to non existing peripheral at 0x80800EE, value 0x16.
13:50:13.3069 [WARNING] sysbus: Read from an unimplemented register Flash:PECR (0x40023C04), returning a value from SVD: 0x7.
13:50:13.3069 [WARNING] sysbus: Write of value 0x7 to an unimplemented register Flash:PECR (0x40023C04) generated from SVD.
13:50:13.3070 [WARNING] sysbus: [cpu: 0x8004A40] ReadDoubleWord from non existing peripheral at 0x80800D8.
13:50:13.3070 [WARNING] sysbus: [cpu: 0x8004A40] ReadDoubleWord from non existing peripheral at 0x80800DC.
13:50:13.3070 [WARNING] sysbus: [cpu: 0x8004A40] ReadDoubleWord from non existing peripheral at 0x80800E0.
13:50:13.3071 [WARNING] sysbus: [cpu: 0x8004A40] ReadDoubleWord from non existing peripheral at 0x80800E4.
13:50:13.3071 [WARNING] sysbus: [cpu: 0x8004A40] ReadDoubleWord from non existing peripheral at 0x80800E8.
13:50:13.3071 [WARNING] sysbus: [cpu: 0x8004A40] ReadDoubleWord from non existing peripheral at 0x80800F0.
13:50:13.3072 [WARNING] sysbus: [cpu: 0x8004A40] ReadDoubleWord from non existing peripheral at 0x80800F4.
13:50:13.3072 [WARNING] sysbus: [cpu: 0x8004A40] ReadDoubleWord from non existing peripheral at 0x80800F8.
13:50:13.3072 [WARNING] sysbus: [cpu: 0x8004A40] ReadDoubleWord from non existing peripheral at 0x80800FC.
13:50:13.4382 [WARNING] spi2: SPI transmission while no SPI peripheral is connected.

```

Image of it
[image]

Renode, version 1.10.1.26379 (b7e0773f-202007301407)
environment, fedora 32, gnome-terminal.

PiotrZierhoffer, 2020-10-06:
Hi @karlp 

We are aware that this happens sometimes, but it's quite difficult to reproduce. We'll try to mitigate this effect, but for now one option would be to edit `~/.config/renode/config` and set the `collapse-repeated-log-entries` to `False`.

This is, obviously, just a workaround, until we find the underlying cause.

PiotrZierhoffer, 2020-10-06:
Did you notice this issue to occur in any specific circumstances? Like window resizing?

karlp, 2020-10-06:
No, I wasn't resizing the window ever, but this was in verbose mode, definitely seemed related to the repeating messages though, it would scroll "normally" when the messages were different and then get stuck somehow on the repeated