renode/renode #130: Add support to NXP LPC43xx microcontroller

evolentini, 2020-09-11:
I am a professor of Computer Engineering at the National University of Tucumán in Argentina, and in our course we want to adopt the use of Renode to develop a part of the laboratories of our asignatures. We are currently using the [EDU-CIAA-NXP](http://www.proyecto-ciaa.com.ar/index_comprar_educiaanxp.html) boards, developed by a consortium between companies and the educational sector in our country. These boards use an NXP LPC4337 as main processor and we want to add support for this processor and this board at Renode.

mgielda, 2020-10-01:
Hi @evolentini this sounds great! You can refer to https://github.com/renode/renode/pull/112 which is a PR for another NXP platform on how to do this. Let us know about your progress and whether we can be of any help.

We will be running some Renode-based education courses locally and there are plans to also potentially do that in the US soon. Happy to see adoption of Renode in academia worldwide!

evolentini, 2020-10-03:
Hi @evolentini I already have a preliminary definition of the CPU, a full implementation of the GPIO ports, and also I have an initial definition of the board. Do not do the pull requests yet because I was waiting for formatting observations and code guidelines on [this](https://github.com/renode/renode-infrastructure/pull/18) pull request in infrastructure repository that will solve some problems in GDB emulation.

PiotrZierhoffer, 2021-05-25:
Hi @evolentini. I was wondering if you had any other code for the board? 

If you have a board description that bundles it all together, we'd be very happy to merge it.

As for now, we have merged the GPIO controller and the display model, as well as the GDB updates. You might be interested to notice that the GDB register access layer has changed significantly and now provides proper XML-based information.