renode/renode #131: Unable to read or write on UART for Contiki NG example

Anshul-99, 2020-09-15:
I'm new to this software and am trying to implement it with Contiki NG. I tried emulating the test-uart example based on Contiki NG for cc2538. The problem is that I'm unable to read or write anything on the uart0 line. The only output I'm getting is the printf statement present in the code. 

I went to the directory where the .resc file exists and entered
`make renode TARGET=cc2538dk SCRIPT=test-uart.resc`

After the renode logger starts up, I switch to the node using 
`mach set "node_1"`

Then I enter 
```
sysbus EnablePeripheral uart0
sysbus.uart0 WriteChar 0x00000020

```
In case my approach is wrong or if I'm using the wrong set of commands, do let me know.

Here is the code in my .resc file
```
:name: cc2538 uart 1 node
$bin=@test-uart.cc2538dk
$name="node_1"
i @scripts/single-node/cc2538.resc
showAnalyzer sysbus.uart0
showAnalyzer sysbus.uart1
mach clear
```

Here is the Contiki NG based code that I'm trying to execute
```
#include "contiki.h"
#include "sys/etimer.h"
#include "dev/leds.h"
#include "dev/uart.h"
#include "dev/serial-line.h"
#include <stdio.h>
#include <stdint.h>
/*---------------------------------------------------------------------------*/
static struct etimer et;
/*---------------------------------------------------------------------------*/
PROCESS(cc2538_uart_demo_process, "cc2538 uart demo");
AUTOSTART_PROCESSES(&cc2538_uart_demo_process);
/*---------------------------------------------------------------------------*/
unsigned int
uart1_send_bytes(const unsigned char *s, unsigned int len)
{
  unsigned int i = 0;

  while(s && *s != 0) {
    if(i >= len) {
      break;
    }
    uart_write_byte(1, *s++);
    i++;
  }
  return i;
}
/*---------------------------------------------------------------------------*/
PROCESS_THREAD(cc2538_uart_demo_process, ev, data)
{
  char *rxdata;
  static uint16_t count=0;
  char string[20];

  //uart_init(1);
  
  PROCESS_BEGIN();
  uart_set_input(1, serial_line_input_byte);
  etimer_set(&et, CLOCK_SECOND * 4);
  leds_toggle(LEDS_GREEN);

  while(1) {
    PROCESS_WAIT_UNTIL(etimer_expired(&et));
    sprintf(string, "sending string %u.\n", ++count);
    uart1_send_bytes((uint8_t *)string, sizeof(string) - 1);
    etimer_reset(&et);
    printf("string is being sent\n");

    PROCESS_YIELD();
    if(ev == serial_line_event_message) {
      leds_toggle(LEDS_RED);
      rxdata = data;
      printf("Data received over UART %s\n", rxdata);
    }
  }

  PROCESS_END();
}
```
[image]

This is the picture of the output I'm getting.

PiotrZierhoffer, 2020-09-16:
Hi @Anshul-99 

A few remarks:

- `sysbus EnablePeripheral` is not needed in your case. Peripherals are enabled by default.
- writing to UART should be as easy as typing on your keyboard in the relevant window, no need to use `WriteChar` from the Monitor
- your code explicitly does not call `uart1_init` - are you sure the UART is enabled in Contiki?
- did you have a chance to verify this on hardware?
- we'd need more data to tell you why are these things happening. It would make a lot of sense if you ran these commands and attach the log file:

```
logFile $CWD/renode-log
sysbus LogPeripheralAccess uart0
sysbus LogPeripheralAccess uart1
cpu LogFunctionNames true
```
Then wait until the application is started and run:
```
log "Begin test"
```
and start typing on your UART.

This will generate a lot of logs, but will help us tracing the problem down.

Anshul-99, 2020-09-17:
Hi @PiotrZierhoffer 

I'm using an example code from the Contiki-NG repo so I presumed there wouldn't be any error in the code. I did check later on and realized that the uart_init() was missing. 

I don't have access to hardware, thus I can't verify it. 

I tried entering data on the uart monitor but it wouldn't accept the data. 

I tried to create a log file in the way you suggested but then the software started to behave in an unexpected manner. After I entered the commands in the log monitor and entered `start` to start the emulation, the program got stuck in some kind of a loop. The node's uart monitor was blank and didn't seem to initialize the program code. I'm attaching the log file for your reference. 

If I didn't enter the log commands, the program behaved as it did earlier but I still couldn't enter any data on the uart monitor.

I think it might be because I'm running the software on a Virtual Machine. I tried increasing the allocated RAM for the VM but it still gave the same output. 

[renode-log.txt](https://github.com/renode/renode/files/5238233/renode-log.txt)

Anshul-99, 2020-11-26:
Hi @PiotrZierhoffer. Any solution regarding this issue?

PiotrZierhoffer, 2021-01-04:
@Anshul-99 

It seems that the problem is with the code itself. CC2538dk provides only one UART line, more on that in the [wiki](https://github.com/contiki-ng/contiki-ng/wiki/Platform-cc2538dk#selecting-uart0-andor-uart1), and by default it is configured to use `uart0`. 

To fix the issue you need to change [board.h](https://github.com/contiki-ng/contiki-ng/blob/release/v4.5/arch/platform/cc2538dk/dev/board.h#L103) and [cc2538-conf.h](https://github.com/contiki-ng/contiki-ng/blob/release/v4.5/arch/cpu/cc2538/cc2538-conf.h#L124).

With these changes we are able to run it successfully in Renode.

The reason why `sysbus.uart0 WriteChar 0x00000020` did not work is that nothing really received the character to echo it back.

```
diff --git a/arch/cpu/cc2538/cc2538-conf.h b/arch/cpu/cc2538/cc2538-conf.h
index d34c3dee5..100940754 100644
--- a/arch/cpu/cc2538/cc2538-conf.h
+++ b/arch/cpu/cc2538/cc2538-conf.h
@@ -123,23 +123,23 @@
 #endif
 
 #ifndef SERIAL_LINE_CONF_UART
-#define SERIAL_LINE_CONF_UART       0 /**< UART to use with serial line */
+#define SERIAL_LINE_CONF_UART       1 /**< UART to use with serial line */
 #endif
 
 #if !SLIP_ARCH_CONF_USB
 #ifndef SLIP_ARCH_CONF_UART
-#define SLIP_ARCH_CONF_UART         0 /**< UART to use with SLIP */
+#define SLIP_ARCH_CONF_UART         1 /**< UART to use with SLIP */
 #endif
 #endif
 
 #if !DBG_CONF_USB
 #ifndef DBG_CONF_UART
-#define DBG_CONF_UART               0 /**< UART to use for debugging */
+#define DBG_CONF_UART               1 /**< UART to use for debugging */
 #endif
 #endif
 
 #ifndef UART1_CONF_UART
-#define UART1_CONF_UART             0 /**< UART to use for examples relying on
+#define UART1_CONF_UART             1 /**< UART to use for examples relying on
                                            the uart1_* API */
 #endif
 
diff --git a/arch/platform/cc2538dk/dev/board.h b/arch/platform/cc2538dk/dev/board.h
index 2540b7ce7..77f414f7f 100644
--- a/arch/platform/cc2538dk/dev/board.h
+++ b/arch/platform/cc2538dk/dev/board.h
@@ -113,11 +113,11 @@
  * UART1_* below.
  * @{
  */
-#define UART0_RX_PORT           GPIO_A_NUM
-#define UART0_RX_PIN            0
+#define UART1_RX_PORT           GPIO_A_NUM
+#define UART1_RX_PIN            0
 
-#define UART0_TX_PORT           GPIO_A_NUM
-#define UART0_TX_PIN            1
+#define UART1_TX_PORT           GPIO_A_NUM
+#define UART1_TX_PIN            1
 
 #define UART1_CTS_PORT          GPIO_B_NUM
 #define UART1_CTS_PIN           0
```

closed 2021-02-11 by PiotrZierhoffer