renode/renode #132: a sample resc file for running zephyr http server over frdmk64f

hakehuang, 2020-09-15:
is it possible to have a sample resc file to demo how to run zephyr http server over frdmk64f?

PiotrZierhoffer, 2020-09-16:
Hi Hake

As we currently don't have any K64F samples, we'll verify if everything works fine with the HTTP setup.

hakehuang, 2020-09-17:
@piotr-binkowski  thanks a lot, I check the test code, and try them, they works fine, just do not know how to connect to a wired network. this would be great if we can have a demo which create two instances and do tcp/udp transactions.

jensschroer, 2020-11-03:
@hakehuang Do you want two instances running in Renode or one in Renode and one outside of Renode?

hakehuang, 2020-11-04:
@jensschroer one in Renode and one outside of Renode?

jensschroer, 2020-11-10:
@hakehuang you would probably want to wait until #147 is merged, as it adds support for the PHY and allows your code to bring up the Ethernet interface automatically.

Having said that, is your host OS Linux? If so, I can provide the steps you need to create a connection between your simulation and the host machine.

PiotrZierhoffer, 2020-11-12:
Hake, Jens, #147 is now merged

jensschroer, 2020-11-13:
> Hake, Jens, #147 is now merged

Thank you!

closed 2021-02-25 by hakehuang