renode/renode #133: Not possible to execute RTOS in STM32

srak2016, 2020-09-29:
Hi, I'm new to renode and just started exploring it. I tried to execute a proprietary RTOS in STM32F4. But it is not working. Looks like the scheduler is not working. Just for information, the same binary is working in QEMU. I would like to know, at present, whether renode can support executing a RTOS in STM32?

mgielda, 2020-10-01:
Hi @srak2016 - there are many RTOSes running on STM32 platforms (and STM32F4 specifically) in Renode. You'd need to explain in more detail what exact platform you are trying to run (not just the CPU family, but rather the SoC model or board) and what the software you are trying to run is doing, what s the log you are getting etc.

srak2016, 2020-10-01:
Hi, I was using STM32F4 Discovery board project.
https://github.com/renode/renode/blob/master/platforms/boards/stm32f4_discovery.repl 

I'm sorry, I have not saved the debug log messages. I'll collect the log and attach here.

karlp, 2020-10-07:
does your alternate rtos use non-systick timers?

srak2016, 2020-10-10:
No, it uses system tick.

I tried to debug and found that the boot process is waiting for RCC_RFCGR: SW bit to be set. 

[image]

I disabled the wait. Now, I can see the RTOS is working. My sample code has a single Task with 1 Second delay.
`void LedTask_1(void)
{
    for (;;) {
       Sleep(1000); /* 1 second */
    }
}`

[image]

And, it is working!! Great!

From log, it seems the sleep is more than 1 second, but I guess it is something to do in RTOS configuration, not related to Renode.

closed 2020-10-10 by srak2016

karlp, 2020-10-10:
ah, you need somethign similar to this then: https://github.com/renode/renode/pull/129

srak2016, 2020-10-11:
Hi karlp, that helps. Thank you!!