renode/renode #139: Compile LiteX / VexRiscv from source and simulate in Renode

lindemer, 2020-10-14:
I would like to simulate a LiteX SoC running VexRiscv with custom CPU plugins. Is there some way to import a LiteX SoC that's been compiled from source into Renode? If so, is there any documentation on how to do this? I could be wrong, but it looks like the emulated hardware Renode uses is downloaded in binary form in the `.resc` scripts.

PiotrZierhoffer, 2020-10-14:
Hi @lindemer 

LiteX hardware is modeled in Renode, and the downloaded binaries are only firmware.
You cannot import the SoC per se, but you can create a platform description that matches your SoC.

Please take a look at https://github.com/litex-hub/litex-renode - it should help you create the needed files.

mateusz-holenko, 2020-10-14:
As for custom CPU plugins, in Renode you can extend the "standard" VexRiscv model by adding new CSRs or instructions.

You can either:

* extend the CPU model directly in C# - here is an example: https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Cores/RiscV/VexRiscv.cs#L59-L61
* or provide the implementation as a python script - take a look at the robot test verifying this functionality: https://github.com/renode/renode/blob/master/tests/unit-tests/riscv-custom-instructions.robot#L58

lindemer, 2020-10-14:
I see, thanks for the information. The project I'm working on requires RISC-V PMP support, but VexRiscv left that feature unimplemented. Does Renode emulate hardware with this level of detail (i.e., is there a 1-1 match between which instructions and CSRs are supported in VexRiscv and in the Renode emulation)? Or is it more generic?

mithro, 2020-10-14:
FYI - @tcal-x

tcal-x, 2020-10-14:
@mateusz-holenko @PiotrZierhoffer , this is a little tangential, but kind of relevant both here and with our discussion this morning.  Could Renode accurately simulate exactly a particular Riscv variant (imacfd subset) even if we gave it that information?   Would it trap / raise an exception if the software contained an instruction that's not implemented?

Ah, I found the `cpuType` attribute: 

https://github.com/renode/renode/blob/8bf570eb3ea908de0511d9e2b1d825c8e420347c/platforms/cpus/litex_vexriscv_linux.repl#L15

So, is Renode pretty strict in this regard?   Say if you issue an atomic instruction and don't have 'a' in your `cpuType`?

PiotrZierhoffer, 2020-10-15:
While we can set specific variants, no all checks are there. We do verify atomics, compressed and floats/doubles, I believe.

PiotrZierhoffer, 2020-10-15:
@lindemer we don't have this level of detail yet, but there is some. For example, VexRiscv by defaults targets Privilege Architecture 1.9, and PMP is supported from 1.10. This can be changed in a repl file.

We do aim to have this feature mapping more precise, but it's not there yet.

lindemer, 2020-10-15:
Okay, thanks for clearing that up :)

closed 2020-10-15 by lindemer