renode/renode #140: Document `emulation` command

xobs, 2020-10-15:
There is a multi-node script (https://github.com/renode/renode/blob/8bf570eb3ea908de0511d9e2b1d825c8e420347c/scripts/multi-node/cc2538/rpl-udp.resc#L12) that uses an undocumented command `emulation`. This command should be documented:

```
(machine-0) help
Available commands:
Name              | Description
================================================================================
alias             : sets an alias.
allowPrivates     : allow private fields and properties manipulation.
analyzers         : shows available analyzers for peripheral.
commandFromHistory: executes command from history.
createPlatform    : creates a platform.
execute           : executes a command or the content of a variable.
help              : prints this help message or info about specified command.
history           : prints command history.
include           : loads a monitor script, python code or a plugin class.
log               : logs messages.
logFile           : sets the output file for logger.
logLevel          : sets logging level for backends.
mach              : list and manipulate machines available in the environment.
macro             : sets a macro.
numbersMode       : sets the way numbers are displayed.
path              : allows modification of internal 'PATH' variable.
pause             : pauses the emulation.
peripherals       : prints list of registered and named peripherals.
python            : executes the provided python command.
quit              : quits the emulator.
require           : verifies the existence of a variable.
runMacro          : executes a command or the content of a macro.
set               : sets a variable.
showAnalyzer      : opens a peripheral backend analyzer.
start             : starts the emulation.
string            : treat given arguments as a single string.
using             : expose a prefix to avoid typing full object names.
verboseMode       : controls the verbosity of the Monitor.
version           : shows version information.
watch             : executes a command periodically, showing output in monitor
You can also provide a device name to access its methods.
Use <TAB> for auto-completion.
(machine-0) help emulation
No such command: emulation.
(machine-0) emulation
The following methods are available:
 - Void AddMachine (Machine machine, String name = "")
 - IMACInterface CreateAndGetTap (String hostInterfaceName, String name, Boolean persistent 
= False)
...
```

xobs, 2020-10-15:
Additionally, what other commands are available that are undocumented?

PiotrZierhoffer, 2020-10-15:
The difference here is that while these commands are static, and actually defined as commands, everything else is autogenerated from code.

These autogenerated things are not documented yet. This involves `emulation`, `machine`, `sysbus` objects, as well as all peripherals in a given machine (e.g. `sysbus.uart`).

There is also the `EmulationManager` object, that has all its methods available directly. The interesting ones are `Save` and `Load` for state saving and loading; and `Clear` to clear the emulation state.

We'd definitely want to have it covered, but we need to come up with an idea of reducing the number of methods available in the Monitor first.

xobs, 2020-10-16:
My view is that being able to network objects together is a big draw of Renode, and having these commands undocumented hampers discoverability.

The current approach for new users seems to be to have them use an existing config as their base and adapt it to their needs. This is adequate for simple designs, but it can rapidly become difficult to scale when there is no close match to base a design off of.

How hard would it be to list every possible verb, even if it's not possible to get `help` on that verb?

PiotrZierhoffer, 2020-10-16:
Correct me if I'm wrong, but I think you want to:

a) have the word `emulation` and all other reserved names listed somewhere, so that people know it's available
b) discover the options available under the these reserved names, without specific help.

This is discoverable right now, in a way: 

- `help` gives you a list of built-in commands
- `peripherals` gives you a list of peripherals for the current machine/context
- `<TAB><TAB>` gives you a list of all available commands, which is the only full list

This, obviously is hardly discoverable.

My idea would be to:
- add a note about `peripherals` to `help`
- add all reserved names to `help` as well, in a small subsection (as they are used in a slightly different way). 

This should at least make things more or less consistent.

To have a proper documentation we'd need, as I said, limit available methods (which is something we're considering, but will most likely not happen very soon).

xobs, 2020-10-16:
Wow, I didn't know `<TAB><TAB>` was a thing. That's amazing. I also didn't know that `peripherals` got promoted to the root namespace -- for example, `sysbus.cpu` appears to be equivalent to `cpu`.

Yes, adding a note about `peripherals` indicating that items under that namespace are also valid commands.

Adding reserved names is also a good idea.

It would be good if most commands appeared under `help` with descriptions of what they are, but I understand that's not easy to do right now. Additionally, commands such as `emulation` have no description, as I gather they're just objects.

So yes, your proposals would go a long way towards resolving this issue.

PiotrZierhoffer, 2020-10-16:
`sysbus.cpu` being equal to `cpu` is not really a thing. There is a command `using` which automatically tries a prefix. Most of our scripts have `using sysbus` in them.