renode/renode #144: Unable to combine paths

basilfx, 2020-10-25:
I want to load a platform description, from a variable:

```
$RIOTBASE=@/Users/basilfx/Desktop/RIOT

machine LoadPlatformDescription $RIOTBASE/cpu/efm32/families/efm32gg/dist/efm32lg990f256.repl
machine LoadPlatformDescription $RIOTBASE/boards/stk3600/dist/stk3600.repl
```

This fails:

```
Could not tokenize here:
machine LoadPlatformDescription /Users/basilfx/Desktop/RIOT/cpu/efm32/families/efm32gg/dist/efm32lg990f256.repl
                                ^
```

I also tried:

```
$RIOTBASE=@/Users/basilfx/Desktop/RIOT

machine LoadPlatformDescription "$RIOTBASE/cpu/efm32/families/efm32gg/dist/efm32lg990f256.repl"
machine LoadPlatformDescription "$RIOTBASE/boards/stk3600/dist/stk3600.repl"
```

But that fails as well (does not expand `$RIOTBASE`):
```
There was an error executing command 'machine LoadPlatformDescription "$RIOTBASE/cpu/efm32/families/efm32gg/dist/efm32lg990f256.repl"'
Could not find file '$RIOTBASE/cpu/efm32/families/efm32gg/dist/efm32lg990f256.repl'.
```

And, I also tried:
```
$RIOTBASE="/Users/basilfx/Desktop/RIOT"

machine LoadPlatformDescription @$RIOTBASE/cpu/efm32/families/efm32gg/dist/efm32lg990f256.repl
machine LoadPlatformDescription @$RIOTBASE/boards/stk3600/dist/stk3600.repl
```

But that fails as well (does not expand `$RIOTBASE`):
```
There was an error executing command 'machine LoadPlatformDescription @$RIOTBASE/cpu/efm32/families/efm32gg/dist/efm32lg990f256.repl'
Could not find file '$RIOTBASE/cpu/efm32/families/efm32gg/dist/efm32lg990f256.repl'.
```

But if I try this:
```
machine LoadPlatformDescription $ORIGIN/cpu/efm32/families/efm32gg/dist/efm32lg990f256.repl
machine LoadPlatformDescription $ORIGIN/boards/stk3600/dist/stk3600.repl
```

It kind of works (the filed does not exist at the given path, but apparently `$ORIGIN` is treated differently).
```
There was an error executing command 'machine LoadPlatformDescription @/Users/basilfx/Desktop/RIOT/boards/stk3600/dist/cpu/efm32/families/efm32gg/dist/efm32lg990f256.repl'
Could not find file '/Users/basilfx/Desktop/RIOT/boards/stk3600/dist/cpu/efm32/families/efm32gg/dist/efm32lg990f256.repl'.
```

How do I define `$RIOTBASE` to behave as `$ORIGIN`? My use-case is that I don't want to load a CPU or board from the default folder, but from a variable (that is actually injected in a script before running Renode).

I'm using Renode version 1.11.0.31461, using macOS 10.15.7.

basilfx, 2020-10-25:
Since `RIOTBASE` is on the environment, my current work-around is:

```
$cpu = `python "print '@' + System.Environment.GetEnvironmentVariable('RIOTBASE') + '/cpu/efm32/families/efm32gg/dist/efm32lg990f256.repl'"`
$board = `python "print '@' + System.Environment.GetEnvironmentVariable('RIOTBASE') + '/boards/stk3600/dist/stk3600.repl'"`

machine LoadPlatformDescription $cpu
machine LoadPlatformDescription $board
```

PiotrZierhoffer, 2020-11-04:
Hey Bas!

This is true. We do have variables like $CWD and $ORIGIN (set when you run a script to its location), and they are concatenable, but for some reason others are not.

This might not be very trivial fix, but we'll look into it.

Citrullin, 2020-11-09:
@basilfx You are kidding me. I work on something for RIOT OS and have the same issue ^^
What are you working on? If it is the same, we maybe should talk :D

Citrullin, 2020-11-09:
The workaround isn't working for me.
```
There was an error executing command 'machine LoadPlatformDescription $board'
Error E00: Syntax error, unexpected '$'; expected end of input
At /home/citrullin/git/RIOT/boards/stm32f7discovery/dist/stm32f7_discovery-bb.repl:1:1:
$cpu = `python "print '@' + System.Environment.GetEnvironmentVariable('RIOTBASE') + '/cpu/stm32f7/dist/stm32f746.repl'"
```

basilfx, 2020-11-09:
@Citrullin You are missing a backtick at the end.

I am busy [generating](https://github.com/basilfx/EFM2RIOT/tree/feature/simplify/dist-development/contrib/renode/platforms) board + CPU definitions for the EFM32 families. When running `make emulate` with Renode as emulator, it loads a [board](https://github.com/basilfx/EFM2RIOT/blob/feature/simplify/dist-development/boards/slstk3301a/dist/board.resc) file that will set-up the emulation. It is then up to the board file to load either from Renode, or from other locations relative to `RIOTBASE`.

Citrullin, 2020-11-09:
> @Citrullin You are missing a backtick at the end.
>
> I am busy [generating](https://github.com/basilfx/EFM2RIOT/tree/feature/simplify/dist-development/contrib/renode/platforms) board + CPU definitions [...]

Maybe I should copy the full message. It is there ^^

Citrullin, 2020-11-09:
I guess the problem is that this kind of nested.
I do this in board.resc
```
$board = `python "print '@' + System.Environment.GetEnvironmentVariable('RIOTBASE') + '/boards/stm32f7discovery/dist/stm32f7_discovery-bb.repl'"`

mach create
using sysbus

machine LoadPlatformDescription $board
```
and this in stm32f7_discovery-bb.repl
```
$cpu = `python "print '@' + System.Environment.GetEnvironmentVariable('RIOTBASE') + '/cpu/stm32f7/dist/stm32f746.repl'"`

using $cpu
```

PiotrZierhoffer, 2020-11-13:
@basilfx @Citrullin it seems that the base problem is fixed. Will be merged when our tests pass and will allow to just

```
$test=@path
i $test/something
```

mgielda, 2020-11-13:
Hi @Citrullin you mention some Discord server, could you point me to it? interested to see your progress. Note we are in progress with integrating with the real Arduino IDE ;) not that it's our main/recommended way to user Renode but we want to enable that for education etc.

Citrullin, 2020-11-13:
> 
> 
> Hi @Citrullin you mention some Discord server, could you point me to it? interested to see your progress. Note we are in progress with integrating with the real Arduino IDE ;) not that it's our main/recommended way to user Renode but we want to enable that for education etc.

Yes, I get that. The Arduino IDE is just a bit limiting for me. Also hear that from other people as well. I started with the Arduino IDE though :D As a lot of people, I guess :) The [discord is one of some bigger coding Youtuber](https://discord.com/invite/Vehs99V). I'm a mod there. Interesting to get in contact with all of these younger Javascript developer. I will probably share the first working version on the [RIOT forum](https://forum.riot-os.org/) first to get some feedback, before I give it to them. 
Of course, one major feature will be renode for the emulations. :)

PiotrZierhoffer, 2020-11-13:
Hey all, the fix is merged on master. Thanks for pointing this out!

closed 2020-11-13 by PiotrZierhoffer

basilfx, 2020-11-16:
Thanks for the feature, I did a first test and it seems to work as expected!