renode/renode #145: Default font for monitor is tiny on high DPI displays

whoopsmith, 2020-10-27:
I see references to default font size for termsharp in ConfigurationManager but its not clear to me how that gets set.

Is there a way to change the font size?

whoopsmith, 2020-10-27:
Update:  I found Ctrl-Shift-+ (on Linux) that increases the font size but not where to set the default.

PiotrZierhoffer, 2020-10-27:
Hi @whoopsmith 

You can set the default font face and size in `~/.config/renode/config`. Look for `font-size` in the `[termsharp]` section.

whoopsmith, 2020-10-27:
@PiotrZierhoffer Thank you. That works bur does not seem to also scale the default terminal size. I don't see a config option for the terminal size.  Is that a configure option?

PiotrZierhoffer, 2020-10-28:
@whoopsmith no, we do not have scaling ability at the moment. 

I can see that adding this as a config option should be relatively trivial, so we'll try to do it soon.

PiotrZierhoffer, 2020-10-28:
These options are now on master. To use them, please compile the latest Renode and run it at least once - the options will be added to your config file as 
```
window-width = 700 
window-height = 400
```

whoopsmith, 2020-10-28:
Thank you.  I've updated to latest master and I can confirm the above works for me.

closed 2020-10-28 by whoopsmith