renode/renode #146: -p flag is not working

mtrzas, 2020-10-29:
Hi,
I tried to use renode with the -p / --plain flag but it doesn't remove the steering code from colours.

In my python script I start renode like this:
"subprocess.Popen(['renode', '-p', '-P', '1234', path_to_resc], stdout=subprocess.PIPE, stderr=subprocess.PIPE)"

When I want to wait for renode to start and completely loaded the renode script I need to wait for:
"(machine-0) \x1b[0m" 
instead of
"(machine-0)" 

I want to remove the colours

mateusz-holenko, 2021-02-24:
Hi @mtrzas, thanks for reporting an issue!

https://github.com/renode/renode/commit/2c5fe97a1eed00d016a970592a9a5533f6784662 should resolve the issue, could you verify if it works for you?

One note: the plain mode works for log messages and monitor output, but we don't strip control codes from UART output.

mtrzas, 2021-03-17:
Now it's working!
Thank you!

closed 2021-03-17 by mtrzas