renode/renode #148: RFC: DHCP Server and improved support for host interaction

jensschroer, 2020-11-03:
I am currently in the process of writing an application that has to interact with a device, for which I do not have the hardware yet. 
Therefore, I am using Renode to simulate the device and communicate with it from my Linux host via the tap interface.
The device will request its IP via DHCP.

For that to work in Linux , I have to

1. Configure a bridge device (create it, allow multicast, give it a static IP)
2. Connect it to the tap the device
3. Configure iptables to allow the traffic
4. Start dnsmasq on the bridge device

I would like to automate these steps from inside Renode but would like to get some feedback from the Renode team first.
Would this make sense?

If so, my approach would be to wrap these steps utilising the sudo framework in Renode (but in its own class), to keep it simple.
Would this be acceptable?

PiotrZierhoffer, 2020-11-05:
Hi Jens

I understand that this procedure is quite tedious.

Our current approach to this is to actually add a server in Renode, so that it can be controlled from within the environment.

We created a modular server infrastructure (https://github.com/renode/renode/tree/master/src/Renode/Network/NetworkServer) and added support for TFTP (https://github.com/renode/renode/blob/master/src/Renode/Network/NetworkServer/Modules/TftpServerModule.cs).

We're also looking at getting the DHCP server in, for example with https://github.com/jpmikkers/DHCPServer/, but this will take some time to get implemented. We'd also like to have some form of DHCP forwarding in it, but, again, this is only a rough plan right now.

If your approach helps then we can certainly take a look. If I understand correctly, the result is that your emulated nodes use the same addressing as your physical network, and that's something that would be very nice to have simplified.

It will, probably, not be very portable between Linux and macOS (not to mention Windows), but it's definitely something we could start with.

jensschroer, 2020-11-10:
Hi Piotr,

Thank you for the response.
I looked at the included modular server infrastructure and also investigated different DHCP libraries for C#. 

However, I came to the conclusion that I would want to run the firmware in an environment that resembles the final system as much as possible, and also would have to use other tools during development and testing that require network connectivity to the device. One example is a remote firmware update using mcumgr .
Therefore, I abandoned the plan to add the DHCP server in Renode and rather searched for an alternative solution.

What I am suggesting is closely related to how libvirt manages the setup of the virtual networks for the managed virtual machines, as I think it resembles the situation I was facing.
libvirt uses dnsmasq as DHCP provider, and three different network modes between guests and host network:

- **isolated**: all off-node traffic is completely blocked
- **nat**: outbound traffic to the LAN is allowed, but MASQUERADED
- **forward**: outbound traffic to the LAN is allowed

(more details here https://libvirt.org/firewall.html)

I did have a quick search for macOS and Windows. 
I think it should be doable on macOS (especially considering that libvirt is also available in host mode on macOS).
However, I think it will be more difficult on Windows, but I am no expert on Windows by all means.
But I do not have access to a Windows or Mac (not mentioning time) to work on these two platforms.

If we can agree that this is something you would be happy to accept as contribution, I would happy to start with it.
My next question would be where to implement this. Should I extend the TAPInterface, or create a new class ?

PiotrZierhoffer, 2020-11-12:
Jens

I think it's a sound approach to have an os-specific, yet useful functionality. We can always improve or make it more generic later.

Regarding the technical side - it's a difficult question. Extending TAPInterface seems to make sense, as opening it requires sudo anyway. It could be a powered-up interface version, with additional capabilities.