renode/renode #149: Reading NVIC ISPR register on STM32F4Discovery

cosmindanielradu19, 2020-11-04:
We are trying to run Tock (https://github.com/tock) using Renode.
TockOS relies on reading the ISPR register for finding out the pending interrupts. It seems that we get the following error:

```
01:33:20.8092 [WARNING] nvic: Unhandled read from offset 0x200.
01:33:20.8092 [WARNING] nvic: Unhandled read from offset 0x4.
```

Looking at your source at (https://github.com/renode/renode-infrastructure/blob/d43be25ceaf0c0ca27ec6a4f2cf27368fd255129/src/Emulator/Cores/Arm-M/NVIC.cs#L71) we found out that the function doesn't allow reading this register. Was this intentional or maybe we are doing something wrong?

tobermory, 2020-12-03:
The ```__NVIC_GetPendingIRQ()``` routine in the CMSIS header ```core_cm3.h``` definitely reads from NVIC->ISPR, so that register is readable.  That same file has the struct definition for the NVIC registers, and ISPR appears at offset 0x100, Your post above looks like it is trying to read from offset 0x200??

tobermory, 2020-12-03:
Hmm, I see now (from same header ```core_cm3.h```) that NVIC_BASE is itself offset 0x100 from SCS_BASE (System Control Space), so perhaps your log output is anchored at  SCS_BASE (which is 0xe000e000),

PiotrZierhoffer, 2020-12-03:
@cosmindanielradu19 we recently pushed some changes that actually allowed us to run Tock in Renode on STM32F4.

Please grab the current master and verify if it works for you

alexandruradovici, 2020-12-07:
Did you actually run Tock using renode? IF so, which board did you use, as STM32F4Discovery is not yet supported by Tock.

WojMel, 2020-12-07:
Yes, We did successfully run Tock on Renode.
Tock was built for `stm32f412gdiscovery` board.
```
$ git clone https://github.com/tock/tock.git
$ git clone https://github.com/tock/libtock-c.git
$ rustup toolchain install $(cat tock/rust-toolchain)
$ rustup target add thumbv7em-none-eabi --toolchain $(cat tock/rust-toolchain)
$ make -C libtock-c/examples/cxx_hello
$ make -C tock/boards/stm32f412gdiscovery/
$ cp tock/target/thumbv7em-none-eabi/release/stm32f412gdiscovery.elf .
$ cp libtock-c/examples/cxx_hello/build/cortex-m4/cortex-m4.tbf .
```

You can reuse `stm32f4_discovery.resc` to launch an app:
```
(monitor) $bin=@stm32f412gdiscovery.elf
(monitor) $app=@cortex-m4.tbf
(monitor) i @scripts/single-node/stm32f4_discovery.resc
(STM32F4_Discovery) showAnalyzer sysbus.uart2
(STM32F4_Discovery) sysbus LoadBinary $app 0x08030000
(STM32F4_Discovery) s
```

On uart2 you should see:
```
Initialization complete. Entering main loop
D2 says hello
D1 says hello
D2 says hello
D1 says hello
```

alexandruradovici, 2020-12-07:
I get a DMA error, did you modify tock?

PiotrZierhoffer, 2021-01-04:
Great! As it works for @alexandruradovici and for us, I will close the issue.

closed 2021-01-04 by PiotrZierhoffer