renode/renode #150: Source code for STM32 Timer Tests?

franc0is, 2020-11-10:
I'm trying to use timers on STM32 but cannot seem to get it to work. I noticed you've got Robot tests checking the functionality, but the firmware for those tests is downloaded as an ELF directly. Would it be possible to get the code for those tests to see what timer features are used so I can reproduce a working example on my side?

Thanks!

Ristovski, 2021-01-25:
I too can't seem to get timers to work - source code for those tests would be greatly appreciated!

Ristovski, 2021-01-26:
@franc0is FWIW, I managed to solve most of my timer issues by loading the NVIC at `0xE000E000` instead of `0xE000E100` as noted in https://github.com/renode/renode/issues/84. It would have been great if this was documented :/

grifcj, 2021-01-26:
Was playing with timers on issue #97

It's not pretty, but this gets TIM2 running and firing interrupt based on STM32CubeMX generated project. I'm trying to get a timing accurate ADC going eventually so my code has more stuff going on than just timer, but first step I think is figuring out a time-accurate timer simulation. Trying to get something like a timer firing at 1 Hz, i.e. once per virtual second.

https://github.com/grifcj/renode-timer-example

Not sure what toolchain/IDE you using, but I'm using ARM GNU RM toolchain.
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

Looking for main, I left it where STM32CubeMX put it, Core/Src/main.c.

Something like below...
```
git clone https://github.com/grifcj/renode-timer-example example
cd example
PATH=/path/to/toolchain:$PATH
make
renode renode.resc
```

If you get timer firing at expected rate based on virtual time, would be interested in details.

grifcj, 2021-01-27:
So the key to the timers appears to be setting their frequency parameters accordingly, which can be done in initial *.repl or later. I had configured the MX cube demo for an 8 MHz external clock, 168 MHz sysclk, and 84 MHz source clk to timer2

```
nvic: IRQControllers.NVIC @ sysbus 0xE000E000
    priorityMask: 0xF0
    systickFrequency: 168000000
    IRQ -> cpu@0
```
```
timer2: Timers.STM32_Timer @ sysbus <0x40000000, +0x400>
    -> nvic@28
    frequency: 84000000
    initialLimit: 0xFFFFFFFF
```

Or like so. 
```
sysbus.timer2 Frequency 84000000
```

All the timers seem to share `LimitTimer` type and they end up getting registered to a table that can be queried via `machine GetClockSourceInfo`. As you can see, timer2 is configured to interrupt every 1 second of virtual time, and it does do so.

[image]

minifake, 2021-12-17:
> So the key to the timers appears to be setting their frequency parameters accordingly, which can be done in initial *.repl or later. I had configured the MX cube demo for an 8 MHz external clock, [...]

@grifcj Thanks for your answer, it's very useful. But I still met some problems about the timer, would you please give some information?

Do you mean the `Event Period` queried via `machine GetClockSourceInfo` is **virtual time** but not **real time**?
What should I do if I want it to be **real time**?

Recently I'm trying to run a UDP demo in renode, the UDP packet transmitted every 1 second in STM32F746 hardware board,  but when running the same ELF in Renod, the period is much longer then 1 second. I had to modify the frequency to a very ridiculous  number(1.15GHz) to  correct the time.  (The frequency of timer6 in real hardware is only 10MHz)
```
timer6: Timers.STM32_Timer @ sysbus <0x40001000, +0x400>
    -> nvic@54
    frequency: 1150000000
    initialLimit: 0xFFFF
```
I also modified the sysclk to 24.84GHz ! which should be only 216MHz.
```
nvic: IRQControllers.NVIC @ sysbus 0xE000E000
    priorityMask: 0xF0
    systickFrequency: 24840000000
    IRQ -> cpu@0
```
[image]

Even though modified such ridiculous number, the time is also not stable.

grifcj, 2022-01-27:
> Do you mean the Event Period queried via machine GetClockSourceInfo is virtual time but not real time?
What should I do if I want it to be real time?

The units are in virtual time.

When you say real time, I believe you are asking for how to get the simulation to execute at same speed as real time. To do this requires that your computer can actually run the simulation this fast. In my experience, this hasn't been typical. I've noticed it's typical (under good conditions) for simulation time to pass 1x to 10x slower than real time. It can be quite a bit slower. Speed will be dependent on what your simulating and the time quantum resolution of the simulation. Generally, simulating more cores at finer time quanta is more computationally expensive. 

I just ran the STM32F4 demo included as part of Renode code base, and it's execute 1x or 0.5x speed of real time on my computer (AMD 2950x). You can see this information with the `emulation GetTimeSourceInfo` comand. There's some docs on the time framework here https://renode.readthedocs.io/en/latest/advanced/time_framework.html

[image]

> I had to modify the frequency to a very ridiculous number(1.15GHz) to correct the time. (The frequency of timer6 in real hardware is only 10MHz)

By modifying the timing values this way, you're configuring the simulation to simulate the timers at a different (much higher) frequency than they would normally execute in real time. It's only speeding them up because now less time simulation time has to pass before the events occur. I can see from the clock info that the event period for timer6 is now only 90 us with that configuration.

> Even though modified such ridiculous number, the time is also not stable.

The ratio between real time and simulated time won't be stable as the simulation executes. You can see above that my instantaneous ratio/load changes every time I query it with `GetTimeSourceInfo`.

> I also modified the sysclk to 24.84GHz ! which should be only 216MHz.

The clock models for the hardware are generally quite simple abstractions of the real thing. There's no constraints and relationships between the hardware models are only as realistic as they're coded to be. You could make a very realistic clock system for a MCU, but it's more typical to just hard-code some settings that get the binary past basic clock configuration and running.

Are you using one of the packaged Renode demos, or an ST demo (i.e. installed with their cube software). If you want to point me to the example you're running, I don't mind running it myself and sharing the performance I observe.