renode/renode #151: Adding a peripheral fails, yet it is added into the platform

AntonKrug, 2020-11-10:
When I run polarfire cpu and I run in the monitor:

```
sysbus WriteByte 0x0 0xff
sysbus ReadByte 0x0
0x00
```

I the read/write failed and the read returned 0x00. It outputs warnings too:

```
14:20:13.6947 [WARNING] sysbus: WriteByte to non existing peripheral at 0x0, value 0xFF.
14:20:33.7017 [WARNING] sysbus: ReadByte from non existing peripheral at 0x0.
```

Which is expected, when invoke this command:

```
machine LoadPlatformDescriptionFromString "begining: Memory.MappedMemory @ sysbus 0x0 { size: 0x100 }"
```

Then I get this exception:
```
There was an error executing command 'machine LoadPlatformDescriptionFromString "be
gining: Memory.MappedMemory @ sysbus 0x0 { size: 0x100 }"'
Error E39: Exception was thrown during registration of 'begining in 'sysbus':
Memory size has to be aligned to guest page size.
```

When I do invoke peripherals
```
peripherals
```

It will not show the 'added' peripheral which makes sense, however now writing to the memory will not produce warnings and even returns values like there would be a valid/working memory: 

```
sysbus WriteByte 0x0 0xff
sysbus ReadByte 0x0
0xFF
```

PiotrZierhoffer, 2020-11-12:
Indeed. What's more, `sysbus WhatPeripheralIsAt 0x0`  reports a memory. Thanks for the report.

AntonKrug, 2020-11-12:
Yes, it looks like the page size alignment is checked too late or something which then will cause a phantom peripheral