renode/renode #152: No such command or device: next_value

Citrullin, 2020-11-11:
My setup might be a bit more complicated. I create a simple to install package for RIOT OS. It includes an installation of renode. You can take a look into the repo of the [snap package](https://github.com/Citrullin/RIOT-OS-IDE). 
You can simply build the package with `snapcraft` and install it with `snap install --devmode riot-os-ide_0.1_amd64.snap`
The improved RIOT repo with the required shell script is available [here](https://github.com/Citrullin/RIOT/tree/renode_improvements). 

Change into the hello word example
`cd examples/hello-world`

In there execute
```
RENODE=/snap/bin/riot-os-ide.renode BOARD=cc2538dk make emulate
```

The output should look like this:
```
citrullin@citrullin-ThinkPad:~/git/RIOT/examples/hello-world$ RENODE=/snap/bin/riot-os-ide.renode BOARD=cc2538dk make emulate
/home/citrullin/git/RIOT/dist/tools/renode/run-renode.sh start
### Starting Renode ###
04:42:08.3827 [WARNING] Couldn't start UI - falling back to telnet mode
04:42:08.8778 [WARNING] Could not find '.renode-root' - falling back to current working directory (/home/citrullin/git/RIOT/examples/hello-world) - might cause problems.
04:42:08.9260 [INFO] Monitor available in telnet mode on port 1234
04:42:09.3490 [INFO] Including script: /home/citrullin/git/RIOT/examples/hello-world/bin/cc2538dk/board.resc
04:42:09.4825 [INFO] System bus created.
04:44:06.1286 [INFO] CC2538: Disposed.
```

When I telnet into renode, I get the following result

```
citrullin@citrullin-ThinkPad:~$ telnet localhost 1234
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Renode, version 1.11.0.24832 (ad7ab75a-202010221310)

(monitor) i @/home/citrullin/git/RIOT/examples/hello-world/bin/cc2538dk/board.resc
No such command or device: next_value
```

The board.resc file is available [here](https://github.com/Citrullin/RIOT/blob/renode_improvements/boards/cc2538dk/dist/board.resc). The variables get replaced before using it. 

Is this the reason of the issue?
```
04:42:08.8778 [WARNING] Could not find '.renode-root' - falling back to current working directory (/home/citrullin/git/RIOT/examples/hello-world) - might cause problems.
```
I can also provide `.renode-root` file. But this would still not really work for me, since renode doesn't respect the relative path, as mentioned in #127.

PiotrZierhoffer, 2020-11-12:
Hi @Citrullin 

`.renode-root` is a file that should point to a place where Renode is installed/downloaded/built. Specifically, it expects to find the path `scripts/monitor.py` in the same directory as `.renode-root` file.

The path with `.renode-root` is one of the directories included in Renode's path, but so is the current working directory.

This `monitor.py` provides the `next_value` command.

I'm not really sure if #127 is applicable here. Renode does not handle CWD well inside PythonPeripheral parameters, but is this required in your specific use case?

PiotrZierhoffer, 2021-05-25:
Looking at your snapcraft.yml, this is clearly not a scenario that we support at the moment - we expect Renode to have one common tree for the binary, scripts etc.

I would suggest putting a runner script in `bin` (a one-liner, really) and the whole tree from the portable package in `opt`.

closed 2021-05-25 by PiotrZierhoffer