renode/renode #153: Issue with synchronizing Systick Timer with Real Time

AhmedLeithy, 2020-11-13:
When using Renode to emulate the STM32F407-discovery kit, we ran into an issue with using systick timer interrupts. We have set the systick timer to interrupt at a rate of one HZ, (by simply setting the Systick->Load value to be equivalent to the SystemCoreClock value through the CMSIS interface).

However we notice that the handler is getting called much faster than the required frequency. We are speculating that this may be an issue of virtual time vs. real time, as using the "currentTime" command shows a mismatch between the two. We experimented with changing the value of the CPU's PerformanceInMips variable and setting it to be equivalent to that of the discovery-kit, but that did not seem to fix the issue.

We are unsure what the issue is or what we should be debugging exactly, since this our first time using Renode. Your input would be greatly appreciated.

closed 2020-11-15 by AhmedLeithy

PiotrZierhoffer, 2020-11-16:
Hi @AhmedLeithy 

The mismatch is expected - depending on the software you're running, the emulation can be either in +/- real time, slower than real time or faster (if your application sleeps a lot and you enable this possibility in Renode).

Unfortunately we cannot guarantee the performance that's 1:1 with the real world.

I see you closed the issue. Why is that? Do you need further explanation on the topic?

AhmedLeithy, 2020-11-16:
Hi @PiotrZierhoffer 

Thank you very much for your response. In that case, it makes more sense to compare performance with regards to virtual time only, without considering real time.

I closed the issue because I received similar advice from someone who has worked with Renode before, and as such thought the issue post was no longer needed. 

Thank you again.

tobermory, 2020-12-03:
Nothing to do with Renode, but beware that the Systick Load register is only 24 bits wide. You cannot get a 1Hz interrupt interval if/when your SystemCoreClock is greater than 16MHz (2 ^ 24). This has bitten me before.