renode/renode #154: Feature request: print current Renode version

basilfx, 2020-11-16:
I am looking for something like `renode -v|--version` that just prints the version (e.g. `1.11.0.41926`) to stdout. This way, I can parse this and check if the user has a certain minimal version of Renode installed.

Currently, such an option does not seem to exist (at least, `renode -h` doesn't show it).

closed 2020-11-18 by RenodeBot

PiotrZierhoffer, 2020-11-18:
Thanks for your contribution, merged to master!

tcal-x, 2020-11-18:
Hi @PiotrZierhoffer , is the version information available from inside the monitor, so that a Renode script can do a similar check?   Hmm, I guess that isn't useful unless there is also either an `assert( )` function, or conditional execution if/then/else.   So maybe it makes sense to do the check from outside  Renode.

pgielda, 2020-11-18:
@tcal-x Inside the monitor there is a command "version" (its been there forever):

`(monitor) version`
`Renode, version 1.11.0.24832 (ad7ab75a-202010221310)`

But obviously being able to check the version directly with an argument without running Renode makes a lot of sense.

You could always access this variable via Python bindings:
`# ./renode -e "python 'Antmicro.Renode.Logging.Logger.Log(LogLevel.Info, emulationManager.VersionString)' ; q"`
`00:15:43.6408 [INFO] Loaded monitor commands from: /mnt/desktop/renode/renode_1.11.0_portable/./scripts/monitor.py`
`00:15:43.8749 [INFO] Renode, version 1.11.0.24832 (ad7ab75a-202010221310)`

But obviously its kinda convoluted.