renode/renode #156: Build from source on OSX: Peripheral not found: uart0

neophob, 2020-11-20:
I use the nrf52840 cpu - I saw that there were recent changes in the unreleased version so I build the OSX version myself. However the `showAnalyzer` command fails now - which worked in the official released version:

```
Renode, version 1.11.0.29015 (8b91fe42-202011201605)
(monitor) mach create
(machine-0) machine LoadPlatformDescription @platforms/cpus/nrf52840.repl
(machine-0) 
(machine-0) $bin?=@/Users/michaelvogt/_code/moskito/watch-renode/nrf52832_xxaa.out
(machine-0) 
(machine-0) showAnalyzer uart0
There was an error executing command 'showAnalyzer uart0'
Received 'Peripheral not found: uart0' error while initializing analyzer for: uart0. Are you missing a required plugin?
(machine-0) 

```

any hints?

PiotrZierhoffer, 2020-11-20:
Hi

All peripherals form a tree and they are to be referred to with a fill name, like `sysbus.uart0`. You can see the whole structure when you type in the `perihperals` command.

But for most of our scripts we add a shortcut for the common `sysbus.` prefix by running `using sysbus` at the beginning of the script. If you run it your command will work.

neophob, 2020-11-20:
doh! thanks

closed 2020-11-20 by neophob