renode/renode #157: Emulator stops when exiting TCP server

aabadie, 2020-11-23:
I'm trying to control an emulated machine using the TCP server that is available with the `-P` option but I exit the telnet session, the emulator is stopped although I think it should remain open.

A simple test sequence:
- in one terminal, run `renode -P 1234`
- in another terminal, connect to the TCP server using `telnet localhost 1234`.

=> In the monitor console, if the `^]` escape character is sent, the telnet session is closed and it also stops the renode emulator. That doesn't seem like a normal behavior, but maybe this is intentional ?

To give you more details, I'm trying to find a way to fully reset a running machine by sending the `machine RequestReset` command to the emulator console via TCP and without keeping the connection opened (so just open the connection, send `machine RequestReset` and close) and this behavior is blocking me.

mateusz-holenko, 2021-05-25:
With https://github.com/renode/renode/commit/81474aaf2748f3d250df6c7f8a7ac3f55d16b976 you should be able to reconnect to Monitor using `telnet`.

In order to close Renode completely, you need to connect and execute the `quit` command.

aabadie, 2021-09-23:
Hi, do you have any plan to release a version which includes this fix ?

dokodokoR, 2022-04-04:
@aabadie 
Release v1.12.0 seems to include https://github.com/renode/renode/commit/81474aaf2748f3d250df6c7f8a7ac3f55d16b976. 

from https://github.com/renode/renode/compare/v1.12.0...master

[image]