renode/renode #158: NRF52 using Softdevice 130 segfaults

neophob, 2020-11-25:
Hi

I use a nrf52 binary compiled with softdevice support (basically out of the box nordic tool). When I use this verbose repl file and load it:

```
using sysbus

mach create
machine LoadPlatformDescription @platforms/cpus/nrf52840.repl

$bin?=@/path/to/bin

showAnalyzer uart0
showAnalyzer uart1

sysbus LogPeripheralAccess uart0
sysbus LogPeripheralAccess uart1
cpu LogFunctionNames true

macro reset
"""
    sysbus LoadELF $bin
"""
runMacro $reset
```

this results in

```
00:38:13.1975 [INFO] cpu: Entering function bsp_board_init at 0x43C
00:38:13.1976 [INFO] cpu: Entering function bsp_board_init at 0x456
00:38:13.1976 [INFO] cpu: Entering function main at 0x11C0
00:38:13.1976 [INFO] cpu: Entering function app_uart_init (entry) at 0x5F4
00:38:13.1976 [INFO] cpu: Entering function app_uart_init at 0x606
00:38:13.1977 [INFO] cpu: Entering function app_fifo_init (entry) at 0x4A8
00:38:13.1978 [INFO] cpu: Entering function app_fifo_init at 0x4AA
00:38:13.1978 [INFO] cpu: Entering function app_fifo_init at 0x4AC
00:38:13.1978 [INFO] cpu: Entering function app_fifo_init at 0x4B2
00:38:13.1978 [INFO] cpu: Entering function app_uart_init at 0x610
00:38:13.1979 [INFO] cpu: Entering function app_uart_init at 0x614
00:38:13.1979 [INFO] cpu: Entering function app_fifo_init (entry) at 0x4A8
00:38:13.1979 [INFO] cpu: Entering function app_fifo_init at 0x4AA
00:38:13.1979 [INFO] cpu: Entering function app_fifo_init at 0x4AC
00:38:13.1980 [INFO] cpu: Entering function app_fifo_init at 0x4B2
00:38:13.1980 [INFO] cpu: Entering function app_uart_init at 0x61E
00:38:13.1980 [INFO] cpu: Entering function app_uart_init at 0x622
00:38:13.1981 [INFO] cpu: Entering function nrf_drv_uart_init (entry) at 0x788
00:38:13.1981 [INFO] cpu: Entering function nrf_drv_uart_init at 0x7C6
00:38:13.1982 [INFO] cpu: Entering function nrfx_uarte_init (entry) at 0xE70
00:38:13.1982 [INFO] cpu: Entering function nrfx_uarte_init at 0xE90
00:38:13.1982 [INFO] cpu: Entering function nrfx_prs_acquire (entry) at 0x810
00:38:13.1982 [INFO] cpu: Entering function nrfx_prs_acquire at 0x81A
00:38:13.1983 [INFO] cpu: Entering function nrfx_uarte_init at 0xE98
00:38:13.1983 [INFO] cpu: Entering function nrfx_uarte_init at 0xEA0
00:38:13.1984 [INFO] cpu: Entering function nrfx_uarte_init at 0xEA6
00:38:13.1984 [WARNING] gpio0: Trying to write pin #6 that is not configured as output
00:38:13.1984 [WARNING] gpio0: Unhandled write to offset 0x718. Unhandled bits: [1] when writing value 0x3. Tags: INPUT (0x1).
00:38:13.1985 [INFO] cpu: Entering function nrfx_uarte_init at 0xECE
00:38:13.1985 [INFO] cpu: Entering function nrfx_uarte_init at 0xF66
00:38:13.1986 [INFO] cpu: Entering function nrfx_uarte_init at 0xF7E
00:38:13.1986 [INFO] cpu: Entering function nrf_uarte_event_clear (entry) at 0xD5C
00:38:13.1986 [INFO] cpu: Entering function nrfx_uarte_init at 0xF8A
00:38:13.1986 [INFO] cpu: Entering function nrf_uarte_event_clear (entry) at 0xD5C
00:38:13.1987 [INFO] cpu: Entering function nrfx_uarte_init at 0xF94
00:38:13.1988 [INFO] cpu: Entering function nrf_uarte_event_clear (entry) at 0xD5C
00:38:13.1988 [INFO] cpu: Entering function nrfx_uarte_init at 0xF9E
00:38:13.1989 [INFO] cpu: Entering function nrf_uarte_event_clear (entry) at 0xD5C
00:38:13.1989 [INFO] cpu: Entering function nrfx_uarte_init at 0xFA8
00:38:13.1989 [INFO] cpu: Entering function nrfx_uarte_init at 0x1000
00:38:13.1989 [INFO] cpu: Entering function nrf_drv_uart_init at 0x7D6
00:38:13.1991 [INFO] cpu: Entering function app_uart_init at 0x66E
00:38:13.1991 [INFO] cpu: Entering function app_uart_init at 0x670
00:38:13.1991 [INFO] cpu: Entering function app_uart_init at 0x67A
00:38:13.1991 [INFO] cpu: Entering function nrf_drv_uart_rx.constprop.0 (entry) at 0x50C
00:38:13.1992 [INFO] cpu: Entering function nrf_drv_uart_rx.constprop.0 at 0x520
00:38:13.1992 [INFO] cpu: Entering function __FLASH_segment_size__+0x300524 (guessed) at 0x400524
00:38:13.1992 [INFO] cpu: Entering function UsageFault_Handler (entry) at 0x364
```

Any hints what's going wrong here, why it fails to load?

neophob, 2020-12-01:
Progress here, I configured the firmware to use the internal oscillator and added the missing softdevice binary to the script:

```
using sysbus

mach create
machine LoadPlatformDescription @platforms/cpus/nrf52832.repl

$bin?=@/fw/pca10040/s132/_build/file.elf
$softdevice?=@/fw/pca10040/s132/s132.bin

logLevel -1
showAnalyzer uart0
showAnalyzer uart1

sysbus LogPeripheralAccess uart0
sysbus LogPeripheralAccess uart1
cpu LogFunctionNames true
sysbus LogAllPeripheralsAccess true

macro reset
"""
    sysbus LoadBinary $softdevice 0x0
    sysbus LoadELF $bin
"""
runMacro $reset

```

Here is the current (verbose) debug output of the application:
```
....
02:21:19.1452 [NOISY] nvic: Disabled IRQ 77.
02:21:19.1452 [NOISY] nvic: Disabled IRQ 78.
02:21:19.1452 [NOISY] nvic: Disabled IRQ 79.
02:21:19.1453 [INFO] cpu: Entering function sd_nvic_critical_region_enter at 0x449FC
02:21:19.1453 [INFO] cpu: Entering function sd_nvic_critical_region_enter at 0x44A02
02:21:19.1454 [INFO] cpu: Entering function __sd_nvic_irq_enable (entry) at 0x44996
02:21:19.1454 [INFO] cpu: Entering function __sd_nvic_irq_enable at 0x4499C
02:21:19.1454 [INFO] cpu: Entering function sd_nvic_critical_region_enter at 0x44A06
02:21:19.1455 [INFO] cpu: Entering function app_util_critical_region_enter at 0x44A8C
02:21:19.1455 [NOISY] cpu: Allocated 64B pointer at 0x140294195579392.
02:21:19.1455 [NOISY] cpu: Allocated is now 98.62MiB.
02:21:19.1455 [NOISY] cpu: Deallocated a 64B pointer at 0x140294195579392.
02:21:19.1456 [INFO] cpu: Entering function nrf_sdh_enable_request at 0x534C8
02:21:19.1456 [NOISY] cpu: Allocated 64B pointer at 0x140295586992272.
02:21:19.1457 [NOISY] cpu: Allocated is now 98.62MiB.
02:21:19.1457 [NOISY] cpu: Deallocated a 64B pointer at 0x140295586992272.
02:21:19.1457 [INFO] cpu: Entering function sd_softdevice_enable (entry) at 0x531E0
02:21:19.1457 [NOISY] nvic: Internal IRQ 11.
02:21:19.1458 [NOISY] cpu: IRQ 0, value True
02:21:19.1458 [NOISY] cpu: Setting CPU IRQ #0 to True
02:21:19.1458 [NOISY] nvic: Acknowledged IRQ 11.
02:21:19.1459 [NOISY] cpu: IRQ 0, value False
02:21:19.1459 [NOISY] cpu: Setting CPU IRQ #0 to False
02:21:19.1459 [NOISY] cpu: Allocated 64B pointer at 0x140294018699152.
02:21:19.1460 [NOISY] cpu: Allocated is now 98.62MiB.
02:21:19.1460 [NOISY] cpu: Loop to itself detected
02:21:19.1461 [NOISY] cpu: Deallocated a 64B pointer at 0x140294018699152.
02:21:19.1461 [INFO] cpu: Entering function SVC_Handler (entry) at 0x262E6
```

If I reduce the error logs to the warning level I see
```
11:35:14.5326 [WARNING] nvic: Unhandled read from offset 0x180.
11:35:14.5326 [INFO] nvic: [cpu: 0x449B4] ReadUInt32 from 0x180 (SysTickValue+0x168), returned 0x0.
11:35:14.5326 [INFO] nvic: [cpu: 0x449B4] WriteUInt32 to 0x180 (SysTickValue+0x168), value 0xBDFF06FC.
11:35:14.5327 [WARNING] nvic: Unhandled read from offset 0x184.
11:35:14.5327 [INFO] nvic: [cpu: 0x449B4] ReadUInt32 from 0x184 (SysTickValue+0x16c), returned 0x0.
11:35:14.5327 [INFO] nvic: [cpu: 0x449B4] WriteUInt32 to 0x184 (SysTickValue+0x16c), value 0xFFFFFFFF.
```

Going down the rabbit hole Register 0x180 looks like the "Interrupt Clear-Enable Registers" (https://developer.arm.com/documentation/ddi0337/e/nested-vectored-interrupt-controller/nvic-programmer-s-model/nvic-register-descriptions) which should be read and writable. Looking at the implementation (https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Cores/Arm-M/NVIC.cs) it looks like the read implementation is missing.

@PiotrZierhoffer in my current understanding the application just stops when entering the SVC_Handler (softdevice). What do you think, could this be related to the missing nvic implementation?

neophob, 2020-12-01:
see https://github.com/renode/renode-infrastructure/pull/23

neophob, 2020-12-02:
I added the test binary, test scripts, verbose logs and detailed instructions here: https://github.com/neophob/renode-nrf52

neophob, 2020-12-03:
Interesting! my nrf52 application needs (as every other app that uses a ble stack) the softdevice binary blob.

I have a combined binary file which includes the app and softdevice file, but I also have my elf application and the softdevice file independent. My expectation was that both options:
- the large, combined binary file 
- the elf and softdevice version loaded at the correct positions 

results in the same output. However this is not the case. If I load the elf and softdevice I do
```
    sysbus LoadBinary $softdevice 0x0
    sysbus LoadELF $binary
```

If I run the combined file I use
```
    sysbus LoadBinary $bin 0x0
    sysbus LoadSymbolsFrom $symb
```

Both options load PC & SP registers correctly, but they start to differ in the init code.

Elf&SoftDevice:
```
14:13:07.6268 [INFO] cpu: Entering function errata_182 (entry) at 0x269BC
14:13:07.6268 [INFO] cpu: Entering function errata_182 at 0x269D4
14:13:07.6268 [INFO] cpu: Entering function SystemInit at 0x26528
14:13:07.6269 [INFO] cpu: Entering function SystemInit at 0x2653A
14:13:07.6269 [INFO] nvic: [cpu: 0x2653A] ReadUInt32 from 0xD88 (CoprocessorAccessControl), returned 0x0.
14:13:07.6269 [INFO] nvic: [cpu: 0x2653A] WriteUInt32 to 0xD88 (CoprocessorAccessControl), value 0xF00000.
14:13:07.6269 [INFO] cpu: Entering function SystemInit at 0x265B2
14:13:07.6270 [INFO] cpu: Entering function SystemInit at 0x265BE
14:13:07.6270 [INFO] cpu: Entering function SystemInit at 0x26622 <<
14:13:07.6270 [INFO] cpu: Entering function SystemCoreClockUpdate (entry) at 0x26320
14:13:07.6271 [INFO] cpu: Entering function SystemInit at 0x26626
14:13:07.6271 [INFO] cpu: Entering function Reset_Handler at 0x262CA
14:13:07.6271 [INFO] cpu: Entering function _start (entry) at 0x26240
14:13:07.6271 [INFO] cpu: Entering function memset (entry) at 0x57C08

```

Combined Binary:
```
14:14:35.6164 [INFO] cpu: Entering function errata_182 (entry) at 0x269BC
14:14:35.6165 [INFO] cpu: Entering function errata_182 at 0x269D4
14:14:35.6165 [INFO] cpu: Entering function SystemInit at 0x26528
14:14:35.6166 [INFO] cpu: Entering function SystemInit at 0x2653A
14:14:35.6166 [INFO] nvic: [cpu: 0x2653A] ReadUInt32 from 0xD88 (CoprocessorAccessControl), returned 0x0.
14:14:35.6166 [INFO] nvic: [cpu: 0x2653A] WriteUInt32 to 0xD88 (CoprocessorAccessControl), value 0xF00000.
14:35.6167 [INFO] cpu: Entering function SystemInit at 0x26566 <<
14:14:35.6167 [WARNING] sysbus: Write of value 0x1 to an unimplemented register NVMC:CONFIG (0x4001E504) generated from SVD.
14:14:35.6168 [WARNING] sysbus: Read from an unimplemented register NVMC:READY (0x4001E400), returning a value from SVD: 0x1.
14:14:35.6168 [INFO] cpu: Entering function SystemInit at 0x2657A
14:14:35.6169 [WARNING] sysbus: Read from an unimplemented register NVMC:READY (0x4001E400), returning a value from SVD: 0x1.
14:14:35.6169 [INFO] cpu: Entering function SystemInit at 0x2659A
14:14:35.6170 [WARNING] sysbus: Write of value 0x0 to an unimplemented register NVMC:CONFIG (0x4001E504) generated from SVD.
14:14:35.6170 [WARNING] sysbus: Read from an unimplemented register NVMC:READY (0x4001E400), returning a value from SVD: 0x1.
14:14:35.6170 [INFO] cpu: Entering function SystemInit at 0x265AE
14:14:35.6171 [INFO] cpu: Entering function NVIC_SystemReset (entry) at 0x262F0
14:14:35.6171 [INFO] nvic: [cpu: 0x262F0] ReadUInt32 from 0xD0C (ApplicationInterruptAndReset), returned 0xFA050000.
14:14:35.6171 [INFO] nvic: [cpu: 0x262F0] WriteUInt32 to 0xD0C (ApplicationInterruptAndReset), value 0x5FA0004.
14:14:35.7332 [INFO] cpu: Entering function NVIC_SystemReset at 0x26310 (2344)
14:14:35.7332 [INFO] machine-0: Machine paused.
<REBOOT>
``` 

@mateusz-holenko is there any reason for that? the application itself is 100% the same... I'm puzzled..

jain0nikhil, 2021-02-19:
@neophob are you able to get it working ? Please share solution if you got one.

neophob, 2021-02-19:
@jain0nikhil - nah unfortunately not yet, I also asked in the Nordic forum. Do you have news or ideas how to resolve this?

jain0nikhil, 2021-02-20:
@PiotrZierhoffer Is there any know reason for this to happening ? Can you share some pointer to resolve this?

Xenoamor, 2022-05-25:
Not sure if any of you guys got any further but the following works for me:

```
using sysbus

mach create "nrf52840"
machine LoadPlatformDescription @platforms/cpus/nrf52832.repl

showAnalyzer uart1

cpu FpuEnabled true

macro reset
"""
    # Load softdevice
    sysbus LoadHEX $ORIGIN/softdevice/s140/hex/s140_nrf52_7.2.0_softdevice.hex
    # Load application
    sysbus LoadELF $ORIGIN/application.elf
"""

runMacro $reset

start
```

You need to comment out the contents of the `SystemInit()` function in system_nrf52840.c as well as there's resets in here which get it stuck in a rebooting loop

Xenoamor, 2023-08-16:
@neophob I found this glorious comment here: https://github.com/renode/renode/issues/5#issuecomment-961715549 which explains why it hangs on the following:
`cpu: Entering function NVIC_SystemReset at 0x26310 (2344)`

Turns out you need to add the following command after having loaded all the binaries:
```sysbus.cpu VectorTableOffset 0x0```

Here's my .resc as it stands:

```
using sysbus

mach create
machine LoadPlatformDescription @nrf52832.repl

macro reset
"""
    sysbus LoadHEX @s132_nrf52_7.2.0_softdevice.hex
    sysbus LoadELF @application.elf

    # Loading binaries will change vector table. Needs to be at 0x0 so soft device handles interrupts
    sysbus.cpu VectorTableOffset 0x0
"""
runMacro $reset
```