renode/renode #161: Consider adding currentCPU#PC info to unhandled read/write logs

kartben, 2020-12-08:
I've found that adding the current value of PC to the various loggers that log unhandled read/writes is super useful to pinpoint where in the emulated code calls are being made to unsupported/unemulated features.
Would be happy to provide a pull request if people are interested.

kartben, 2020-12-08:
just realized there is a utility method `DecorateWithCPUNameAndPC` -- I will reword the title of the issue accordingly

mateusz-holenko, 2021-05-26:
Unfortunately I was not able to reproduce the problem - unhandled read/write logs are decorated with the CPU name and PC when the access it made from the CPU thread (IOW by the simulated software).
Could you give some more context in which you don't observe them?

I added automatic robot tests to verify this in: https://github.com/renode/renode/commit/81be8184e9b2c6d06cbf7147366d1096cc9afc45.